rai-opensource/spot_ros2 #11: ERROR:colcon:colcon build: Duplicate package names not supported:

MizzouRobotics, 2023-01-10:
Hey all! I was noticing while trying to build spot_ros2 that I was getting an error or two on the build. Admittedly, I get hit with a wall of errors, so it's hard to fully sus out, but here's the two errors that stood out below. Any guidance would be greatly appreciated.

```
#0 0.981 ImportError: cannot import name 'compiler_class' from partially initialized module 'distutils.ccompiler' (most likely due to a circular import) (/usr/lib/python3/dist-packages/numpy/distutils/ccompiler.py)
#0 0.984 [0.638s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'python_setup_py' in 'lib/python3/dist-packages/numpy': Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.

...

ERROR:colcon:colcon build: Duplicate package names not supported:
- gmock:
  - opt/ros/humble/src/gmock_vendor
  - usr/src/gmock
- gtest:
  - opt/ros/humble/src/gtest_vendor
  - usr/src/gtest
```
Here's the bulk of the Traceback:
```
#0 0.981 Traceback (most recent call last):
#0 0.981   File "<string>", line 1, in <module>
#0 0.981   File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 10, in <module>
#0 0.981     import distutils.core
#0 0.981   File "/usr/lib/python3/dist-packages/numpy/distutils/__init__.py", line 24, in <module>
#0 0.981     from . import ccompiler
#0 0.981   File "/usr/lib/python3/dist-packages/numpy/distutils/ccompiler.py", line 9, in <module>
#0 0.981     from distutils.ccompiler import (

#0 0.984 Traceback (most recent call last):
#0 0.984   File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 142, in _identify
#0 0.984     retval = extension.identify(_reused_descriptor_instance)
#0 0.984   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 48, in identify
#0 0.984     config = get_setup_information(setup_py)
#0 0.984   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
#0 0.984     _setup_information_cache[hashable_env] = _get_setup_information(
#0 0.984   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 281, in _get_setup_information
#0 0.984     result = subprocess.run(
#0 0.984   File "/usr/lib/python3.10/subprocess.py", line 524, in run
#0 0.984     raise CalledProcessError(retcode, process.args,
#0 0.984 subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
#0 0.984 
#0 1.099 Traceback (most recent call last):
#0 1.099   File "<string>", line 1, in <module>
#0 1.099   File "/usr/lib/python3.10/distutils/core.py", line 225, in run_setup
#0 1.099     raise RuntimeError(("'distutils.core.setup()' was never called -- "
#0 1.099 RuntimeError: 'distutils.core.setup()' was never called -- perhaps 'setup.py' is not a Distutils setup script?
#0 1.110 [0.765s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'python_setup_py' in 'lib/python3/dist-packages/numpy/compat': Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
#0 1.110 Traceback (most recent call last):
#0 1.110   File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 142, in _identify
#0 1.110     retval = extension.identify(_reused_descriptor_instance)
#0 1.110   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 48, in identify
#0 1.110     config = get_setup_information(setup_py)
#0 1.110   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
#0 1.110     _setup_information_cache[hashable_env] = _get_setup_information(
#0 1.110   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 281, in _get_setup_information
#0 1.110     result = subprocess.run(
#0 1.110   File "/usr/lib/python3.10/subprocess.py", line 524, in run
#0 1.110     raise CalledProcessError(retcode, process.args,
#0 1.110 subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
#0 1.110 
#0 1.275 Traceback (most recent call last):
#0 1.275   File "<string>", line 1, in <module>
#0 1.275   File "/usr/lib/python3.10/distutils/core.py", line 225, in run_setup
#0 1.275     raise RuntimeError(("'distutils.core.setup()' was never called -- "
#0 1.275 RuntimeError: 'distutils.core.setup()' was never called -- perhaps 'setup.py' is not a Distutils setup script?
#0 1.293 [0.947s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'python_setup_py' in 'lib/python3/dist-packages/numpy/core': Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
#0 1.293 Traceback (most recent call last):
#0 1.293   File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 142, in _identify
#0 1.293     retval = extension.identify(_reused_descriptor_instance)
[... cut ...] s    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
#0 3.932 
#0 16.07 [15.721s] ERROR:colcon:colcon build: Duplicate package names not supported:
#0 16.07 - gmock:
#0 16.07   - opt/ros/humble/src/gmock_vendor
#0 16.07   - usr/src/gmock
#0 16.07 - gtest:
#0 16.07   - opt/ros/humble/src/gtest_vendor
#0 16.07   - usr/src/gtest
```

MizzouRobotics, 2023-01-10:
As another note, I've also tested the incominb jlb/bdai_changes branch, and it doesn't seem to make a difference.

jiuguangw, 2023-01-12:
Hi - I think your problem is duplicate GTest as indicated by this message:
`ERROR:colcon:colcon build: Duplicate package names not supported:
- gmock:
  - opt/ros/humble/src/gmock_vendor
  - usr/src/gmock
- gtest:
  - opt/ros/humble/src/gtest_vendor
  - usr/src/gtest`

ROS comes with its own gmock and gtest. I think if you uninstall / delete the ones located in usr/src, this will go away. Might be a worth a try.

MizzouRobotics, 2023-01-13:
Thanks for the reply! I have a few things that have cropped up, but I'll be able to test this next thursday.

MizzouRobotics, 2023-02-22:
I looked into to this further and after burning hours found it was an issue with me not initializing a global arg in the dockerfile's container context >.<
Thanks!

closed 2023-02-22 by MizzouRobotics

Phyremaster, 2023-11-14:
@MizzouRobotics Sorry to ping you out of nowhere, but I'm having the same issue with a different set of ROS packages and this is the only relevant result I've found on the Internet. Would you please share what the global arg was that fixed this?

rosiakpiotr, 2025-08-27:
> [@MizzouRobotics](https://github.com/MizzouRobotics) Sorry to ping you out of nowhere, but I'm having the same issue with a different set of ROS packages and this is the only relevant result I've found on the Internet. Would you please share what the global arg was that fixed this?

I faced the same issue while running colcon build inside docker container on RaspberryPi with command 
```
docker run --rm -it -v "$PWD":/ws ros:jazzy-ros-base colcon build --symlink-install
```
The issue was, by default container starts in root path of filesystem "/" and the issue was gone after explicit change directory to actual workspace directory rendering the following command:
```
docker run --rm -it -v "$PWD":/ws ros:jazzy-ros-base bash -c "cd /ws && colcon build --symlink-install"
```
After that my workspace built without those errors:
```
ERROR:colcon:colcon build: Duplicate package names not supported:
- gmock:
  - opt/ros/humble/src/gmock_vendor
  - usr/src/gmock
- gtest:
  - opt/ros/humble/src/gtest_vendor
  - usr/src/gtest
```