rai-opensource/spot_ros2 #15: Missing named_arm_positions conversion

benaxelrod, 2023-01-25:
I am trying to stow the arm like this:

```
stow_command = RobotCommandBuilder.arm_stow_command()
sync_command = RobotCommandBuilder.build_synchro_command(stow_command)
action_goal = RobotCommand.Goal()
conv.convert_proto_to_bosdyn_msgs_robot_command(sync_command, action_goal.command)
```

But unfortunately, it looks like there is something missing in the conversion library:

```
  File "/bdai/projects/arm_examples/arm_examples/arm_simple_ros_sac.py", line 345, in stow
    conv.convert_proto_to_bosdyn_msgs_robot_command(sync_command, action_goal.command)
  File "/bdai/ws/build/spot_driver/spot_driver/conversions.py", line 4532, in convert_proto_to_bosdyn_msgs_robot_command
    convert_proto_to_bosdyn_msgs_robot_command_one_of_command(proto, ros_msg.command)
  File "/bdai/ws/build/spot_driver/spot_driver/conversions.py", line 4522, in convert_proto_to_bosdyn_msgs_robot_command_one_of_command
    convert_proto_to_bosdyn_msgs_synchronized_command_request(proto.synchronized_command, ros_msg.synchronized_command)
  File "/bdai/ws/build/spot_driver/spot_driver/conversions.py", line 6615, in convert_proto_to_bosdyn_msgs_synchronized_command_request
    convert_proto_to_bosdyn_msgs_arm_command_request(proto.arm_command, ros_msg.arm_command)
  File "/bdai/ws/build/spot_driver/spot_driver/conversions.py", line 4937, in convert_proto_to_bosdyn_msgs_arm_command_request
    convert_proto_to_bosdyn_msgs_arm_command_request_one_of_command(proto, ros_msg.command)
  File "/bdai/ws/build/spot_driver/spot_driver/conversions.py", line 4900, in convert_proto_to_bosdyn_msgs_arm_command_request_one_of_command
    convert_proto_to_bosdyn_msgs_named_arm_positions_command_request(proto.named_arm_position_command, ros_msg.named_arm_position_command)
  File "/bdai/ws/build/spot_driver/spot_driver/conversions.py", line 5078, in convert_proto_to_bosdyn_msgs_named_arm_positions_command_request
    convert_proto_to_bosdyn_msgs_named_arm_positions_command_positions(proto.position, ros_msg.position)
NameError: name 'convert_proto_to_bosdyn_msgs_named_arm_positions_command_positions' is not defined
```

`conversions.py` is autogenerated, and I'm not sure from what, so I'm not sure how to proceed.

jbarry-bdai, 2023-02-01:
Hm... yeah the auto-generation didn't recognize `NamedArmCommand.Positions` as an enum...

jbarry-bdai, 2023-02-03:
This will be fixed by #18 or by https://github.com/bdaiinstitute/bdai/pull/86 internal to BDAI.

benaxelrod, 2023-02-08:
tested, and fixed.

closed 2023-02-08 by benaxelrod