rai-opensource/spot_ros2 #22: Is this driver ready to use out of the box? Arm control?

dcolli23, 2023-02-14:
Hello!

Our lab purchased a Spot recently and we're trying to setup ROS on Spot using the Boston Dynamics Core as the machine running roscore. I've got a few questions about this repository if someone has some time:

- Is this ROS driver in a usable state out of the box? 
- What Spot SDK versions are supported?
- Does the repository offer full Spot arm control? 
- Is there any Spot Python API functionality that is missing from this driver?

We'd love to use this repo but it would require a good amount of ROS1->ROS2 updating for the lab's repos so we're trying to weigh our options before diving in.

Thank you!

jbarry-bdai, 2023-02-15:
* It _should_ be ready to use out of the box but I'm not aware of anyone outside MASKOR and BDAI using it yet.  We'd love your feedback!
* We support 3.2 and are working on updating to 3.3 with a beta copy
* Currently we support everything that goes through the `RobotCommand` proto including the arm.
* Many pieces of functionality are still missing - essentially everything that does not go through `RobotCommand` (e.g. autowalk, door opening, etc).  [bosdyn_msgs](https://github.com/bdaiinstitute/bosdyn_msgs) should be a 1-1 conversion of all the BD API messages to ROS messages and the `conversions.py` file in here contains the auto-generated python code to convert back and forth but we haven't written the backend Python code to support it all yet.

This is being supported by MASKOR and BDAI, not by Boston Dynamics themselves.  Spot support will not be able to answer questions about ROS - just want to make that clear!

dcolli23, 2023-02-20:
Thanks so much! This is super helpful. I'll let you know if we have any feedback!

closed 2023-02-20 by dcolli23