rai-opensource/spot_ros2 #33: Dock/undock and other services?

julianraheema, 2023-04-03:
In spot_ros recently added the dock/undock and other services. Is there a plan to do so in spot_ros2? Also the spot_ros we launch `roslaunch spot_vis view_robot.launch` to interactive with robot, and use services? I am not sure how to do that with spot_ros2.

jbarry-bdai, 2023-04-04:
We're actually in the process of combining the back end for this with the ROS1 wrapper so it will be easier to add those services.  We don't have interactive markers or anything set up in ROS2 yet.

julianraheema, 2023-04-04:
@jbarry-bdai I appreciate your feedback, and please point me to the new stuff on ROS2 once are ready. I think it will be a good idea to migrate these services and features from ros1 to ros2. The sooner the better :) and Thank you for your great work!

jbarry-bdai, 2023-04-04:
That just went in!  @heuristicus can you point us at the ROS1 code?  Then I can look at how it needs to change for ROS2 and put up a few instructions :)

heuristicus, 2023-04-04:
It's basically all of the stuff in [spot_viz](https://github.com/heuristicus/spot_ros/tree/master/spot_viz). There are some rviz configurations for panel layouts, with corresponding launch files to view the robot model (useful when checking your urdf extensions), and view_robot will show you some basic output when the driver is running.

The `resource` directory contains a `.ui` file which is loaded by the spot panel when it starts. This file is made using Qt designer and saves a ton of messing around with GUI stuff directly in the code (and there is still a lot of messing around).

The panel itself is in the src directory. I don't know how plugins work in ros2, but in ros1 it is necessary to have a `plugin_description.xml` which is referred to in the package xml with

```
  <export>
      <rviz plugin="${prefix}/plugin_description.xml"/>
  </export>
```

The panel itself just interacts directly with the services and topics made available by the robot.

julianraheema, 2023-04-11:
Thank you very much @jbarry-bdai and @heuristicus I will try to try to integrate the spot_viz to ros2, but there is something that I need to take care of using ROS1 first which is the navigation stack. If you guys know a good reference for the navigation stack for SPOT, I would appreciate it. I was able to create a map using rtabmap, and I think I am now at the point of integrating the navigation stack. Once I am done with this then I can go back and migrate to ROS2. Again thank you for your support and great work.

zmk5, 2023-06-14:
Is the BDAII group open to others implementing the Dock/Undocking services into `spot_ros2` and submitting a merge request? I'm about to implement it myself but wasn't sure if this project is like the `spot-sdk` where contributions are not really addressed.

jbarry-bdai, 2023-06-16:
Yes please - implement away and put up a merge request outlining what you've done and the testing you've run when you're ready!  Thank you!

zmk5, 2023-06-16:
> Yes please - implement away and put up a merge request outlining what you've done and the testing you've run when you're ready! Thank you!

That's great to hear! I'll start implementing the dock/undock services as soon as I can. Does the group have a board showing what features need to be added to take up future tasking?

jbarry-bdai, 2023-06-16:
> That's great to hear! I'll start implementing the dock/undock services as soon as I can. Does the group have a board showing what features need to be added to take up future tasking?

We don't have a board yet but are probably going to be starting one soon.  Meantime feel free to pick up any issue!

JacobJBeck, 2024-02-06:
I believe this issue was resolved by https://github.com/bdaiinstitute/spot_ros2/pull/91 and can be closed. From my testing, the dock service at least works.

julianraheema, 2024-02-06:
I appreciate the hard work. Thank you.

closed 2024-02-06 by julianraheema

zmk5, 2024-02-11:
Anytime. Let me know when the team has a board so I can contribute more to the project!