rai-opensource/spot_ros2 #43: No model displayed in rviz

Jaynkd, 2023-04-24:
[image]
[image]

I followed the script in repo to install ros2 humble on a docker container and ran the following command:
ros2 launch spot_description description.launch.py

I got the blank/black screen in rviz.

Jaynkd, 2023-05-02:
Thanks @davidwatkins-bdai for the reply. Yes, i figured out it is a problem on the docker side. I got the robot working in my laptop with everything installed on my system.
Question: is robot supposed to update or move in real time when the real spot robot moves in real env. I see the fiducial markers getting updated in real time with lag but not the movement of the robot. I can see the joint states updating though through ros2 topics but in rviz it doesn't reflec anything or any movement in regard to the spot.

davidwatkins-bdai, 2023-05-02:
The RobotModel visualization should be able to show `/joint_states` updates in RViz. I would recommend double-checking that it is pointing to the correct topic. We have had some trouble getting multiple Docker containers to talk to each other without using Cyclone DDS if you have one container publishing and another doing visualization. If they are all part of the same container then it is likely that your RViz is pointed to the wrong topic.

Jaynkd, 2023-05-05:
I still have problems with docker diisplaying rviz but it works on my laptop with humble installed.

Jaynkd, 2023-05-05:
I have a followup question was the spot_description launch file ever used for real spot robot testing. I tried using spot_description launch file with real spot to visualize real spot in rviz using the command:
ros2 launch spot_description description.launch.py gui:=False
It does not or work partially. I see the frames moving but not the spot model inside the rviz.

The scenario is I am walking the robot with spot controller in the real time. I want to visualize the spot robot movement in rviz. I did launched the spot driver, sourced the local install in ros and then ran the above mentioned ros2 command for spot description. I did configured the spot_driver to use with real spot.

I did notice if I close the joint state publisher gui. The spot model in rviz does reflect the true real spot behavior or movements happening in real environment. I can legs moving in respect to real spot but why it is not working with the argument gui:=False in following command.

ros2 launch spot_description description.launch.py gui:=False

davidwatkins-bdai, 2023-05-07:
We don't use `description.launch,py` to load the model into RViz. Please look at `spot_ros2/spot_driver/launch/spot_driver_with_namespace.launch.py` on L17 to see how we use it. If the launch file is not operating correctly then you can always open a PR to try and address it or create a separate issue specifically indicating the error you are having. 

If you are just trying to visualize the Spot in realtime, look at the `spot_driver` launch file I mentioned. We use that and are able to visualize the robot in RViz with no issues.

Jaynkd, 2023-05-08:
@davidwatkins-bdai how or what args
1) I can use for rviz2 with real spot.
2) I can use for rviz2 (or using joint state publisher gui) with simulated spot

davidwatkins-bdai, 2023-05-08:
`ros2 launch spot_driver spot_driver_with_namespace.launch.py spot_name:=<spot_name> config_file:=/path/to/spot_ros.yaml`

Jaynkd, 2023-05-08:
Will the above ros launch command work for simulated spot too. How to use spot model with arm (simulation) with joint state publisher gui with above launch.

ros2 launch spot_driver spot_driver_with_namespace.launch.py spot_name:=<spot_name> config_file:=/path/to/spot_ros.yaml

Thank you the real spot does work for me. I can see, in rviz2, spot body, spot body and spot arm TFs but not spot arm model. I did tried changing urdf arg in launch file, i do see the arm model with spot arm's TFs but the model doesn't seem connected to spot arm TFs since it does not move with REAL spot arm movement, only TFs for spot arm moves.

[image]

davidwatkins-bdai, 2023-05-08:
I believe arm support is something we are still working on. I believe there may be an argument for a simulated Spot, but I am less familiar with that as I have only tested with the real ones. I believe @jbarry-bdai or @METEORITENMAX could weigh in on the specifics of simulated better than I can. 

Regardless, the driver launch file should give you enough information for writing your own launch file specific to your situation. If you need arm support you likely will need to modify the joint state publisher and everywhere the URDF is mentioned for the arm. I don't believe we have MoveIt! integration working yet, either.

jbarry-bdai, 2023-05-10:
How are you simulating spot?  The driver is looking for the protobuf API messages that Spot publishes so if your simulation publishes those then it should in theory work...

For spots with arms, you should be able to see the arm frames in rviz and those should move correctly.  We have #21 for getting the correct description loaded based on whether the robot has an arm or not but haven't actually implemented that...

Jaynkd, 2023-05-11:
To just see the spot model right with arm I am using ros2 launch spot_description description.py to launch rviz2 to visalize spot body model. It does show spot body and TFs for spot body and spot arm but spot arm model is not seen in that.

I played around and change the the urdf to spot_with_arm.urdf.xacro. It does import spot body and its TFs correctly but  when it comes to spot arm it only shows spot arm TFs and the spot model seems disconnected from TFs as shown in the screnshot above https://github.com/bdaiinstitute/spot_ros2/issues/43#issuecomment-1538777326.

jbarry-bdai, 2023-05-12:
I think you need to change the wrapper to publish the description with the arm on the description topic is my guess?  We haven't tried this yet...

kaiyu-zheng, 2023-06-12:
Curious, with the latest spot_ros2, are you still able to get TF frames for the arm joints @Jaynkd 

I am not able to...
[image]

khughes-bdai, 2024-03-27:
As the problem with the URDF correctly loading has been resolved, and when rviz from the driver is launched, all frames (body and arm) are displayed and move correctly, I'm marking this issue as closed.

closed 2024-03-27 by khughes-bdai