rai-opensource/spot_ros2 #44: Wrong output format of a function in ros_helpers.py

gial11, 2023-04-28:
Hello, it seems that in [this](https://github.com/bdaiinstitute/spot_ros2/blob/ad3e7c1c086c05c1be6018cc285055635e224f3e/spot_driver/spot_driver/ros_helpers.py#L198) function the output definition is wrong. Instead of `Tuple[Image | CompressedImage, CameraInfo]` it should be `Tuple[CompressedImage, CameraInfo]` otherwise the driver fails to start.

gial11, 2023-04-28:
Thank you very much for the detailed explanation. So, we are running the driver on a docker with ros foxy and if we keep `Tuple[Image | CompressedImage, CameraInfo]` the driver fails to start with the following error:

[image]

If we replace with `Tuple[CompressedImage, CameraInfo]` we do not get the images (which now we know it is normal since `CompressedImage` msgs are not currently handled); however if we replace with `Tuple[Image, CameraInfo]` the driver launches correctly.

amessing-bdai, 2023-04-28:
What version of python are you using?

gial11, 2023-04-29:
3.8.10

amessing-bdai, 2023-05-01:
Ok, so the `|` operator for type hints was added in 3.10. I have added changing the `|` to a `Union` which should fix this issue in a PR (#45 ) with some other stuff. Please try running that branch and see if it works for you

closed 2023-05-01 by amessing-bdai