rai-opensource/spot_ros2 #49: Odometry messages published at a low rate

gial11, 2023-05-04:
Hello, odometry messages seem to be published at a low rate (around 3.5 Hz) in the `/odometry` topic. I tried to increase the rate of `robot_state` in both the config file (spot_ros.yaml) and [here](https://github.com/bdaiinstitute/spot_ros2/blob/00bee893d42ab84e09423f5aa401e2162b328278/spot_driver/spot_driver/spot_ros2.py#L1032) without success. I also tried to increase the rate of the whole node [here](https://github.com/bdaiinstitute/spot_ros2/blob/00bee893d42ab84e09423f5aa401e2162b328278/spot_driver/spot_driver/spot_ros2.py#L1027).

Do you have the same behaviour?

gial11, 2023-05-15:
Nevermind, it is an issue related to another package we are using.

skpawar1305, 2023-05-25:
try
https://github.com/bdaiinstitute/spot_ros2/pull/63
give 100 to async_tasks_rate

gial11, 2023-06-02:
Thank you. By doing so, odometry messages are now published at a rate of 10 Hz.

closed 2023-06-02 by gial11