rai-opensource/spot_ros2 #50: Images not obtained for publishing

gial11, 2023-05-04:
Hello, few seconds after having successfully launched the driver I repeatedly receive the following error:<p></p>
`[spot_ros2-1] The following exception was never retrieved: 'NoneType' object has no attribute 'frontleft'`.<p></p>
It seems that the images are not correctly obtained in the functions responsible of publishing them like [this](https://github.com/bdaiinstitute/spot_ros2/blob/00bee893d42ab84e09423f5aa401e2162b328278/spot_driver/spot_driver/spot_ros2.py#L210) one.

skpawar1305, 2023-05-25:
https://github.com/bdaiinstitute/spot_ros2/issues/64#issuecomment-1563285185

gial11, 2023-06-02:
Solved by doing [this](https://github.com/bdaiinstitute/spot_ros2/issues/64#issuecomment-1563285185). Thank you.

closed 2023-06-02 by gial11