rai-opensource/spot_ros2 #64: Driver crashes due to UnsupportedPixelFormatRequestedError

JacobJBeck, 2023-05-25:
After connecting an Estop and starting the driver, I get this error almost immediately:

``` bash
[spot_ros2-1] [ERROR] [1685020675.151819478] [spot_wrapper]: UnsupportedPixelFormatRequestedError. Likely pixel_format is set wrong for some image request
[spot_ros2-1] Spinning ros2_driver
[spot_ros2-1] Traceback (most recent call last):
[spot_ros2-1]   File "/home/jacob/workspace/ros2_humble/install/spot_driver/lib/spot_driver/spot_ros2", line 33, in <module>
[spot_ros2-1]     sys.exit(load_entry_point('spot-driver', 'console_scripts', 'spot_ros2')())
[spot_ros2-1]   File "/home/jacob/workspace/ros2_humble/build/spot_driver/spot_driver/spot_ros2.py", line 1309, in main
[spot_ros2-1]     executor.spin()
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 279, in spin
[spot_ros2-1]     self.spin_once()
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 764, in spin_once
[spot_ros2-1]     self._spin_once_impl(timeout_sec)
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 761, in _spin_once_impl
[spot_ros2-1]     future.result()
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 94, in result
[spot_ros2-1]     raise self.exception()
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[spot_ros2-1]     self._handler.send(None)
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[spot_ros2-1]     await call_coroutine(entity, arg)
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 332, in _execute_timer
[spot_ros2-1]     await await_or_execute(tmr.callback)
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[spot_ros2-1]     return callback(*args)
[spot_ros2-1]   File "/home/jacob/workspace/ros2_humble/build/spot_driver/spot_driver/spot_ros2.py", line 216, in publish_camera_images_callback
[spot_ros2-1]     for image_entry in result:
[spot_ros2-1] TypeError: 'NoneType' object is not iterable
[ERROR] [spot_ros2-1]: process has died [pid 186987, exit code 1, cmd '/home/jacob/workspace/ros2_humble/install/spot_driver/lib/spot_driver/spot_ros2 --ros-args -r __node:=spot_ros2'].
```

I had a quick look, and it seems this is caused at [wrapper line 2612](https://github.com/bdaiinstitute/spot_wrapper/blob/bdacec5bace54f5f4882660bfbe0b2821dfa4855/spot_wrapper/wrapper.py#L2612) which returns `None` when `UnsupportedPixelFormatRequestedError` is caught. Not sure why this is thrown, as pixel format appear to be [set correctly on line 802](https://github.com/bdaiinstitute/spot_wrapper/blob/bdacec5bace54f5f4882660bfbe0b2821dfa4855/spot_wrapper/wrapper.py#L802).

Possibly related to https://github.com/bdaiinstitute/spot_ros2/issues/50

skpawar1305, 2023-05-25:
does your spot has mono cameras?
try this
https://github.com/bdaiinstitute/spot_ros2/pull/61
with
https://github.com/bdaiinstitute/spot_wrapper/pull/15

@heuristicus

AngelRodriguez8008, 2023-05-29:
Hi @skpawar1305

I have the same error with mono cams. I merged in my fork the two spot_mono_cams branches from spot_ros2 & spot_wrapper. But I'm still getting the same error. Is some config required? If yes, where? I have no config file, still running the tutorials with default values  (in Docker).

Thanks for your help.

skpawar1305, 2023-05-29:
> Hi @skpawar1305
> 
> I have the same error with mono cams. I merged in my fork the two spot_mono_cams branches from spot_ros2 & spot_wrapper. But I'm still getting the same error. Is some config required? If yes, where? I have no config file, still running the tutorials with default values  (in Docker).
> 
> Thanks for your help.
> 
> 
> <pre># /opt/ros2_ws# ros2 launch spot_driver spot_driver.launch.py
> [INFO] [launch]: All log files can be found below /root/.ros/log/2023-05-29-19-11-43-130415-SpotCORE-2324
> [INFO] [launch]: Default logging verbosity is set to INFO
> [WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
> [INFO] [spot_ros2-1]: process started with pid [2325]
> [INFO] [robot_state_publisher-2]: process started with pid [2327]
> [robot_state_publisher-2] [INFO] [1685387503.373149263] [robot_state_publisher]: got segment base_link
> [robot_state_publisher-2] [INFO] [1685387503.373294958] [robot_state_publisher]: got segment body
> [robot_state_publisher-2] [INFO] [1685387503.373308507] [robot_state_publisher]: got segment front_left_hip
> [robot_state_publisher-2] [INFO] [1685387503.373316168] [robot_state_publisher]: got segment front_left_lower_leg
> [robot_state_publisher-2] [INFO] [1685387503.373323533] [robot_state_publisher]: got segment front_left_upper_leg
> [robot_state_publisher-2] [INFO] [1685387503.373329476] [robot_state_publisher]: got segment front_rail
> [robot_state_publisher-2] [INFO] [1685387503.373335479] [robot_state_publisher]: got segment front_right_hip
> [robot_state_publisher-2] [INFO] [1685387503.373341734] [robot_state_publisher]: got segment front_right_lower_leg
> [robot_state_publisher-2] [INFO] [1685387503.373347938] [robot_state_publisher]: got segment front_right_upper_leg
> [robot_state_publisher-2] [INFO] [1685387503.373353908] [robot_state_publisher]: got segment rear_left_hip
> [robot_state_publisher-2] [INFO] [1685387503.373360004] [robot_state_publisher]: got segment rear_left_lower_leg
> [robot_state_publisher-2] [INFO] [1685387503.373366364] [robot_state_publisher]: got segment rear_left_upper_leg
> [robot_state_publisher-2] [INFO] [1685387503.373372486] [robot_state_publisher]: got segment rear_rail
> [robot_state_publisher-2] [INFO] [1685387503.373378389] [robot_state_publisher]: got segment rear_right_hip
> [robot_state_publisher-2] [INFO] [1685387503.373384247] [robot_state_publisher]: got segment rear_right_lower_leg
> [robot_state_publisher-2] [INFO] [1685387503.373391088] [robot_state_publisher]: got segment rear_right_upper_leg
> [spot_ros2-1] [INFO] [1685387504.685016275] [spot_ros2]: Starting ROS driver for Spot
> [spot_ros2-1] [INFO] [1685387504.686645870] [spot_wrapper]: Initialising robot at 192.168.50.3
> [spot_ros2-1] [INFO] [1685387504.687260166] [spot_wrapper]: Trying to authenticate with robot...
> [spot_ros2-1] [INFO] [1685387505.362363459] [spot_wrapper]: Successfully authenticated.
> [spot_ros2-1] [INFO] [1685387505.557103772] [spot_wrapper]: Creating clients...
> [spot_ros2-1] <font color="#4E9A06">Hi from spot_driver.</font>
> [spot_ros2-1] <font color="#C4A000">Waiting for estop to be released.  Make sure you have an active estop.  You can acquire an estop on the tablet by choosing &quot;Acquire Cut Motor Power Authority&quot; in the dropdown menu from the power icon.  (This will not power the motors or take the lease.)</font>
> [spot_ros2-1] Found estop!
> [spot_ros2-1] [ERROR] [1685387530.310567150] [spot_wrapper]: UnsupportedPixelFormatRequestedError. Likely pixel_format is set wrong for some image request
> [spot_ros2-1] Spinning ros2_driver
> [spot_ros2-1] Traceback (most recent call last):
> [spot_ros2-1]   File &quot;/opt/ros2_ws/install/spot_driver/lib/spot_driver/spot_ros2&quot;, line 33, in &lt;module&gt;
> [spot_ros2-1]     sys.exit(load_entry_point(&apos;spot-driver&apos;, &apos;console_scripts&apos;, &apos;spot_ros2&apos;)())
> [spot_ros2-1]   File &quot;/opt/ros2_ws/build/spot_driver/spot_driver/spot_ros2.py&quot;, line 1311, in main
> [spot_ros2-1]     executor.spin()
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py&quot;, line 279, in spin
> [spot_ros2-1]     self.spin_once()
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py&quot;, line 764, in spin_once
> [spot_ros2-1]     self._spin_once_impl(timeout_sec)
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py&quot;, line 761, in _spin_once_impl
> [spot_ros2-1]     future.result()
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py&quot;, line 94, in result
> [spot_ros2-1]     raise self.exception()
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py&quot;, line 239, in __call__
> [spot_ros2-1]     self._handler.send(None)
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py&quot;, line 418, in handler
> [spot_ros2-1]     await call_coroutine(entity, arg)
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py&quot;, line 332, in _execute_timer
> [spot_ros2-1]     await await_or_execute(tmr.callback)
> [spot_ros2-1]   File &quot;/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py&quot;, line 107, in await_or_execute
> [spot_ros2-1]     return callback(*args)
[... cut ...] e 216, in publish_camera_images_callback
> [spot_ros2-1]     for image_entry in result:
> [spot_ros2-1] TypeError: &apos;NoneType&apos; object is not iterable
> [ERROR] [spot_ros2-1]: process has died [pid 2325, exit code 1, cmd &apos;/opt/ros2_ws/install/spot_driver/lib/spot_driver/spot_ros2 --ros-args -r __node:=spot_ros2&apos;].
> </pre>

you need to set
https://github.com/bdaiinstitute/spot_ros2/blob/spot_mono_cams/spot_driver/config/spot_ros_example.yaml#L21
to false and use the config file

AngelRodriguez8008, 2023-05-29:
Hi @skpawar1305

It is working. Thank you very much!

> you need to set https://github.com/bdaiinstitute/spot_ros2/blob/spot_mono_cams/spot_driver/config/spot_ros_example.yaml#L21 to false and use the config file with ros2 launch spot_driver spot_driver.launch.py config_file:=

JacobJBeck, 2023-08-22:
I'm still experiencing this error even after setting `rgb_cameras: False` in my config file.

JacobJBeck, 2023-08-22:
The issue seems to be inside `SpotWrapper`, which takes a `rgb_cameras` arg, but [doesn't pass it to `SpotImages`](https://github.com/bdaiinstitute/spot_wrapper/blob/1a04e5da2c9b4213d5d3d9c710e5f5443317f75b/spot_wrapper/wrapper.py#L660). By adding `self._rgb_cameras` to the initializer on this line, I was able to resolve the issue.

heuristicus, 2023-08-22:
Thanks for catching this. I've added a PR which fixes the rgb cameras flag not being passed.

closed 2023-08-23 by bhung-bdai