rai-opensource/spot_ros2 #67: Stricter Code Formatting

philipsgithub, 2023-05-30:
Hi, I would like to propose to define black as default code formatter. I observed that sometimes lines get very long, imports are unordered, function calls are not very comprehensible at first sight, etc. code format checks could also be integrated into the CI pipeline. I think the project would benefit from stricter formatting rules.

note that, it does not have to be black. it can be any code formatter. But I plan to contribute more in the upcoming future and I am missing some style guidelines:)

davidwatkins-bdai, 2023-05-30:
Internally we are using `ruff` and `black`. @ggoretkin-bdai and I have put this infrastructure together. I don't have any problem migrating this over here, but I am unsure how expedient we could be in implementing this. Here is the configuration we are using internally:
```toml
[tool.ruff]
# Enable pycodestyle (`E`), Pyflakes (`F`), and import sorting (`I`)
select = ["E", "F", "I"]
ignore = []

# Allow autofix for all enabled rules (when `--fix`) is provided.
fixable = ["A", "B", "C", "D", "E", "F", "G", "I", "N", "Q", "S", "T", "W", "ANN", "ARG", "BLE", "COM", "DJ", "DTZ", "EM", "ERA", "EXE", "FBT", "ICN", "INP", "ISC", "NPY", "PD", "PGH", "PIE", "PL", "PT", "PTH", "PYI", "RET", "RSE", "RUF", "SIM", "SLF", "TCH", "TID", "TRY", "UP", "YTT"]
unfixable = []

# Exclude a variety of commonly ignored directories.
exclude = [
    ".bzr",
    ".direnv",
    ".eggs",
    ".git",
    ".hg",
    ".mypy_cache",
    ".nox",
    ".pants.d",
    ".pytype",
    ".ruff_cache",
    ".svn",
    ".tox",
    ".venv",
    "__pypackages__",
    "_build",
    "buck-out",
    "build",
    "dist",
    "node_modules",
    "venv",
    "docker/ros",
]

# Same as Black.
line-length = 120

# Allow unused variables when underscore-prefixed.
dummy-variable-rgx = "^(_+|(_+[a-zA-Z0-9_]*[a-zA-Z0-9]+?))$"

# Assume Python 3.10.
target-version = "py310"

[tool.ruff.per-file-ignores]
"__init__.py" = ["F401"]

[tool.ruff.mccabe]
max-complexity = 10

[tool.black]
line-length = 120
target-version = ['py310']
include = '\.pyi?$'
force-exclude = ''''

preview = true
```

philipsgithub, 2023-05-30:
oh nice thanks for providing me with your config. I will have a look at it, since I am not familiar with ruff. but it seems to be an improved `pylint`... we could keep this issue open, as a thread to discuss the next steps together

davidwatkins-bdai, 2023-05-30:
Additionally, we use `pre-commit` to run these automatically. Here is the config we use for this:
```yaml
# Copyright [2023] Boston Dynamics AI Institute, Inc.

repos:
- repo: https://github.com/charliermarsh/ruff-pre-commit
  # Ruff version.
  rev: 'v0.0.263'
  hooks:
    - id: ruff
      args: ['--fix', '--config', 'pyproject.toml'] # we want this to refer to `bdai/pyproject.toml`
- repo: https://github.com/psf/black
  rev: 23.3.0
  hooks:
    - id: black
      language_version: python3.10
      args: ['--config', 'pyproject.toml'] # we want this to refer to `bdai/pyproject.toml`
      verbose: true
- repo: https://github.com/pre-commit/pre-commit-hooks
  rev: v4.4.0
  hooks:
  -   id: check-yaml
  -   id: check-added-large-files
  -   id: check-toml
  -   id: end-of-file-fixer
```

ggoretkin-bdai, 2023-05-31:
I am happy we are using `ruff`. I want to be clear that there are features in `pylint` that are not in `ruff`: https://github.com/charliermarsh/ruff/issues/970

Skylion007, 2023-06-05:
ruff is actually more of an improved flake8 with all its plugins implemented. It now also has support for rules in other linters like pylint. I think they are also working on getting black's rules fully supported natively.

philipsgithub, 2023-06-09:
with the merged #72 this issue can be closed

closed 2023-06-09 by philipsgithub