rai-opensource/spot_ros2 #87: Examples seem to be broken

rshu-bd, 2023-06-14:
Both the [simple_walk_forwad](https://github.com/bdaiinstitute/spot_ros2/tree/main/examples/simple_walk_forward) and [simple_arm_motion](https://github.com/bdaiinstitute/spot_ros2/tree/main/examples/simple_arm_motion) examples seem to be broken. 

The `simple_walk_forward` example fails with 

```
ModuleNotFoundError: No module named 'bdai_ros2_wrappers.action_client_wrapper
```
It tries to import `bdai_ros2_wrappers.action_client_wrapper`
https://github.com/bdaiinstitute/spot_ros2/blob/cca2cb7e68c6b326f44821e6acb0368ca5ad7cd2/examples/simple_walk_forward/simple_walk_forward/walk_forward.py#L2
which seems to have changed to just `bdai_ros2_wrappers.action_client`. 

Also, missing the `node_name` input when initializing the `ActionClientWrapper`
https://github.com/bdaiinstitute/spot_ros2/blob/cca2cb7e68c6b326f44821e6acb0368ca5ad7cd2/examples/simple_walk_forward/simple_walk_forward/walk_forward.py#L25

These changes are sufficient to run the example succesffuly.

amessing-bdai, 2023-06-14:
This is being fixed in #77, so hopefully will be repaired in the near future.

jbarry-bdai, 2023-06-21:
Also #92 sorry!

closed 2023-07-20 by amessing-bdai