rai-opensource/spot_ros2 #96: Colcon Build Failed for spot_msgs

Shehroseraza, 2023-06-27:
--- stderr: spot_msgs                           
CMake Error in CMakeLists.txt:
  Imported target "bosdyn_msgs::bosdyn_msgs__rosidl_generator_py" includes
  non-existent path

    "/usr/local/lib/python3.10/dist-packages/numpy/core/include"
Failed   <<< spot_msgs [10.2s, exited with code 1]

Ubuntu 22.04
Ros Humble 
Installed msgs from source as-well but still not figuring out what to do?

amessing-bdai, 2023-06-28:
Do you have numpy installed?

Shehroseraza, 2023-06-28:
Yes, I also check that Numpy is already to newest version.

amessing-bdai, 2023-06-28:
And it says it is in `/usr/local/lib/python3.10/dist-packages` or somewhere else (e.g. `/home/<username>/.local/lib/python3.10/site-packages`?

Shehroseraza, 2023-06-28:
shehrose@lambda:~/catkin_ros2$ pip3 show numpy
Name: numpy
Version: 1.25.0
Summary: Fundamental package for array computing in Python
Home-page: https://www.numpy.org
Author: Travis E. Oliphant et al.
Author-email: 
License: BSD-3-Clause
Location: /home/shehrose/.local/lib/python3.10/site-packages
Requires: 
Required-by: bosdyn-client

Shehroseraza, 2023-06-28:
PATH=/opt/ros/humble/bin:/home/shehrose/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
GDMSESSION=ubuntu
its already there

Shehroseraza, 2023-06-28:
--- stderr: spot_msgs                          
CMake Error in CMakeLists.txt:
  Imported target "bosdyn_msgs::bosdyn_msgs__rosidl_generator_py" includes
  non-existent path

    "/usr/local/lib/python3.10/dist-packages/numpy/core/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< spot_msgs [9.98s, exited with code 1]
complete error during build

DavidWatkins, 2023-06-30:
Did you install numpy via apt? It sounds like you may not have installed ros2-humble-desktop-full which would have included this. Try installing python3-numpy via apt.

svillemure, 2023-07-17:
I have been getting the same error, @Shehroseraza did you mange to solve the issue?

Shehroseraza, 2023-07-22:
no, i shifted back to ros1 @svillemure

svillemure, 2023-07-24:
I managed to fix this issue by linking the contents of ~/.local/lib/python3.10/site-packages/ to /usr/local/lib/python3.10/dist-packages/ with the command:
```shell
Sudo ln -fs ~/.local/lib/python3.10/site-packages/* /usr/local/lib/python3.10/dist-packages/
```

kahleeeb3, 2023-08-03:
This is suddenly an issue for me as well

kahleeeb3, 2023-08-08:
Running the following worked for me
```
sudo pip install --upgrade --no-deps --force-reinstall numpy
```

skpawar1305, 2023-08-22:
> Did you install numpy via apt? It sounds like you may not have installed ros2-humble-desktop-full which would have included this. Try installing python3-numpy via apt.

most probably the one who generated bosdyn_msgs should do this

ardiya, 2023-10-25:
I have the same problem. `dpkg -L python3-numpy` shows that it's installed in `/usr/lib/python3/dist-packages` instead of `/usr/local/lib/python3/dist-packages`. You can link it like how svillemure did.

stefie100-bdai, 2024-01-22:
I have the same problem.  The pip3 command fixed it for me.

RiccardoZuppetti, 2024-02-20:
> I have the same problem. The pip3 command fixed it for me.

Hi @stefie100-bdai, which command did you use to solve the encountered issue?

closed 2024-06-12 by tcappellari-bdai