rai-opensource/spot_ros2 #100: ModuleNotFoundError: No module named 'bdai_ros2_wrappers'

hermannsblum, 2023-06-30:
When installing according to the README (exact docker build is here: https://github.com/cvg/spot_ros2/blob/main/Dockerfile) and trying to launch the driver, I get the following error:

```
ros2 launch spot_driver spot_driver.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-06-30-11-40-09-971049-a9cd31b36588-2812
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [spot_ros2-1]: process started with pid [2813]
[INFO] [robot_state_publisher-2]: process started with pid [2815]
[robot_state_publisher-2] [INFO] [1688125210.064818506] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1688125210.064890388] [robot_state_publisher]: got segment body
[robot_state_publisher-2] [INFO] [1688125210.064897292] [robot_state_publisher]: got segment front_left_hip
[robot_state_publisher-2] [INFO] [1688125210.064901026] [robot_state_publisher]: got segment front_left_lower_leg
[robot_state_publisher-2] [INFO] [1688125210.064904499] [robot_state_publisher]: got segment front_left_upper_leg
[robot_state_publisher-2] [INFO] [1688125210.064907747] [robot_state_publisher]: got segment front_rail
[robot_state_publisher-2] [INFO] [1688125210.064910865] [robot_state_publisher]: got segment front_right_hip
[robot_state_publisher-2] [INFO] [1688125210.064913930] [robot_state_publisher]: got segment front_right_lower_leg
[robot_state_publisher-2] [INFO] [1688125210.064917021] [robot_state_publisher]: got segment front_right_upper_leg
[robot_state_publisher-2] [INFO] [1688125210.064920023] [robot_state_publisher]: got segment rear_left_hip
[robot_state_publisher-2] [INFO] [1688125210.064923050] [robot_state_publisher]: got segment rear_left_lower_leg
[robot_state_publisher-2] [INFO] [1688125210.064926107] [robot_state_publisher]: got segment rear_left_upper_leg
[robot_state_publisher-2] [INFO] [1688125210.064929089] [robot_state_publisher]: got segment rear_rail
[robot_state_publisher-2] [INFO] [1688125210.064932092] [robot_state_publisher]: got segment rear_right_hip
[robot_state_publisher-2] [INFO] [1688125210.064935084] [robot_state_publisher]: got segment rear_right_lower_leg
[robot_state_publisher-2] [INFO] [1688125210.064938164] [robot_state_publisher]: got segment rear_right_upper_leg
[spot_ros2-1] Traceback (most recent call last):
[spot_ros2-1]   File "/ros2_ws/src/install/spot_driver/lib/spot_driver/spot_ros2", line 33, in <module>
[spot_ros2-1]     sys.exit(load_entry_point('spot-driver==0.0.0', 'console_scripts', 'spot_ros2')())
[spot_ros2-1]   File "/ros2_ws/src/install/spot_driver/lib/spot_driver/spot_ros2", line 25, in importlib_load_entry_point
[spot_ros2-1]     return next(matches).load()
[spot_ros2-1]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[spot_ros2-1]     module = import_module(match.group('module'))
[spot_ros2-1]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[spot_ros2-1]     return _bootstrap._gcd_import(name[level:], package, level)
[spot_ros2-1]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[spot_ros2-1]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[spot_ros2-1]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[spot_ros2-1]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[spot_ros2-1]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[spot_ros2-1]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[spot_ros2-1]   File "/ros2_ws/src/build/spot_driver/spot_driver/spot_ros2.py", line 15, in <module>
[spot_ros2-1]     from bdai_ros2_wrappers.single_goal_action_server import (
[spot_ros2-1] ModuleNotFoundError: No module named 'bdai_ros2_wrappers'
```

I cannot find this package anywhere...

hermannsblum, 2023-07-05:
any update anybody?

davidwatkins-bdai, 2023-07-05:
It looks like you didn't do the `git submodule update`. That dependency is pulled in when you properly initialized `ros_utilities`.

closed 2023-07-31 by amessing-bdai

hermannsblum, 2023-08-15:
@amessing-bdai sorry, took me a while to get back to this, but I can now confirm that this fixed it. Thanks for the pointer :)