rai-opensource/spot_ros2 #119: Robot does not power on

mrayh001, 2023-08-03:
Hi,

While running the ros2 run simple_walk_forward walk_forward example. I see the error 

**[ERROR] [1691096838.474175676] [walk_forward]: Unable to power on robot message was Exception while powering on: bosdyn.api.PowerCommandResponse (ResponseError): Code: 12 (Protobuf definition mismatch?)**

ros2: Humble
ubuntu; 22.04

bhung-bdai, 2023-08-04:
We'll check this out and get back to you. Did you try all the typical steps like power cycling it, rebuilding the driver, reconnecting, etc? Do you have the output of the driver before running the `ros2` example?

mrayh001, 2023-08-04:
This is the output of the drive:
ros2 launch spot_driver spot_driver.launch.py
[INFO] [launch]: All log files can be found below /home/munim/.ros/log/2023-08-04-13-18-39-807579-munim-HP-Z240-Tower-Workstation-11631
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [spot_ros2-1]: process started with pid [11632]
[INFO] [robot_state_publisher-2]: process started with pid [11634]
[robot_state_publisher-2] [INFO] [1691169520.040656217] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1691169520.040790554] [robot_state_publisher]: got segment body
[robot_state_publisher-2] [INFO] [1691169520.040801779] [robot_state_publisher]: got segment front_left_hip
[robot_state_publisher-2] [INFO] [1691169520.040808743] [robot_state_publisher]: got segment front_left_lower_leg
[robot_state_publisher-2] [INFO] [1691169520.040814947] [robot_state_publisher]: got segment front_left_upper_leg
[robot_state_publisher-2] [INFO] [1691169520.040820810] [robot_state_publisher]: got segment front_rail
[robot_state_publisher-2] [INFO] [1691169520.040826619] [robot_state_publisher]: got segment front_right_hip
[robot_state_publisher-2] [INFO] [1691169520.040832345] [robot_state_publisher]: got segment front_right_lower_leg
[robot_state_publisher-2] [INFO] [1691169520.040838156] [robot_state_publisher]: got segment front_right_upper_leg
[robot_state_publisher-2] [INFO] [1691169520.040844051] [robot_state_publisher]: got segment rear_left_hip
[robot_state_publisher-2] [INFO] [1691169520.040849738] [robot_state_publisher]: got segment rear_left_lower_leg
[robot_state_publisher-2] [INFO] [1691169520.040855451] [robot_state_publisher]: got segment rear_left_upper_leg
[robot_state_publisher-2] [INFO] [1691169520.040861305] [robot_state_publisher]: got segment rear_rail
[robot_state_publisher-2] [INFO] [1691169520.040867128] [robot_state_publisher]: got segment rear_right_hip
[robot_state_publisher-2] [INFO] [1691169520.040872942] [robot_state_publisher]: got segment rear_right_lower_leg
[robot_state_publisher-2] [INFO] [1691169520.040878959] [robot_state_publisher]: got segment rear_right_upper_leg
[spot_ros2-1] [INFO] [1691169522.532887231] [spot_ros2]: Hi from spot_driver.
[spot_ros2-1] [INFO] [1691169522.544538663] [spot_ros2]: Starting ROS driver for Spot
[spot_ros2-1] [INFO] [1691169522.546128130] [spot_wrapper]: Initialising robot at 192.168.80.3
[spot_ros2-1] [INFO] [1691169522.546672471] [spot_wrapper]: Trying to authenticate with robot...
[spot_ros2-1] [INFO] [1691169523.251616893] [spot_wrapper]: Successfully authenticated.
[spot_ros2-1] [INFO] [1691169523.518475776] [spot_wrapper]: Creating clients...
[spot_ros2-1] [INFO] [1691169523.679247090] [spot_wrapper]: Choreography is not available.
[spot_ros2-1] [INFO] [1691169523.707314736] [spot_wrapper]: No velodyne point cloud service is available.
[spot_ros2-1] [INFO] [1691169523.717800326] [spot_cam_wrapper]: Trying to authenticate with robot...
[spot_ros2-1] [INFO] [1691169524.656192823] [spot_cam_wrapper]: Successfully authenticated.
[spot_ros2-1] [INFO] [1691169525.090966951] [spot_ros2]: Found estop!
[spot_ros2-1] [INFO] [1691169525.091651760] [spot_ros2]: Spinning ros2_driver
[spot_ros2-1] [INFO] [1691169525.115529351] [spot_ros2]: Driver successfully started!
[spot_ros2-1] [INFO] [1691169707.170649779] [spot_wrapper]: Powering on
[spot_ros2-1] [INFO] [1691169707.240851039] [spot_ros2]: Received goal request
[spot_ros2-1] [ERROR] [1691169707.266749431] [spot_wrapper]: Unable to execute robot command: bosdyn.api.RobotCommandResponse (NotPoweredOnError): The robot must be powered on to accept a command.
[spot_ros2-1] [ERROR] [1691169707.268658811] [spot_ros2]: Error raised in execute callback: bosdyn.api.RobotCommandResponse (NotPoweredOnError): The robot must be powered on to accept a command.
[spot_ros2-1] Traceback (most recent call last):
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/action/server.py", line 325, in _execute_goal
[spot_ros2-1]     execute_result = await await_or_execute(execute_callback, goal_handle)
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[spot_ros2-1]     return callback(*args)
[spot_ros2-1]   File "/home/munim/spotros2_ws/build/spot_driver/spot_driver/spot_ros2.py", line 1582, in handle_robot_command
[spot_ros2-1]     raise Exception(err_msg)
[spot_ros2-1] Exception: bosdyn.api.RobotCommandResponse (NotPoweredOnError): The robot must be powered on to accept a command.
[spot_ros2-1] [WARN] [1691169707.270121612] [spot_ros2]: Goal state not set, assuming aborted. Goal ID: [152  31 163  30  63  57  69 121 138 212 145 154 116   6  61 190]

I did the mentioned solutions. No Luck.

mrayh001, 2023-08-15:
spot_ros2 git commit hash 3d7224cc6d8bcc6724d3cb387b60e93125b205ce  
spot_wrapper: 1a04e5da2c9b4213d5d3d9c710e5f5443317f75b

Sorry for the late reply. I deleted my previous workspace and try to follow the instruction again. Now it does not build the workspace for the following error:

colcon build --symlink-install
Starting >>> spot_msgs
Starting >>> bdai_ros2_wrappers
Starting >>> simple_arm_motion
Starting >>> simple_walk_forward
--- stderr: simple_arm_motion                                                                                                                 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 105, in build
    return await extension.build(additional_hooks=additional_hooks)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/build.py", line 81, in build
    available_commands = await self._get_available_commands(
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/build.py", line 171, in _get_available_commands
    output = await check_output(
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 128, in check_output
    assert not rc, f'Expected {args} to pass: {stderr_data}'
AssertionError: Expected ['/usr/bin/python3', 'setup.py', '--help-commands'] to pass: Traceback (most recent call last):
  File "/home/munim/spotros2_ws/src/spot_ros2/examples/simple_arm_motion/setup.py", line 5, in <module>
    setup(
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 144, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.10/distutils/core.py", line 134, in setup
    ok = dist.parse_command_line()
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/dist.py", line 707, in parse_command_line
    result = _Distribution.parse_command_line(self)
  File "/usr/lib/python3.10/distutils/dist.py", line 480, in parse_command_line
    if self.handle_display_options(option_order):
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/dist.py", line 1092, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
  File "/usr/lib/python3.10/distutils/dist.py", line 683, in handle_display_options
    self.print_commands()
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/dist.py", line 843, in print_commands
    cmdclass = ep.resolve()
  File "/home/munim/.local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/command/bdist_wininst.py", line 1, in <module>
    import distutils.command.bdist_wininst as orig
ModuleNotFoundError: No module named 'distutils.command.bdist_wininst'

---
Failed   <<< simple_arm_motion [1.31s, exited with code 1]
Aborted  <<< bdai_ros2_wrappers [1.32s]
Aborted  <<< simple_walk_forward [1.32s]
Aborted  <<< spot_msgs [8.70s]                           

Summary: 0 packages finished [9.26s]
  1 package failed: simple_arm_motion
  3 packages aborted: bdai_ros2_wrappers simple_walk_forward spot_msgs
  1 package had stderr output: simple_arm_motion
  4 packages not processed
Expected ['/usr/bin/python3', 'setup.py', '--help-commands'] to pass: Traceback (most recent call last):
  File "/home/munim/spotros2_ws/src/spot_ros2/examples/simple_arm_motion/setup.py", line 5, in <module>
    setup(
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 144, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.10/distutils/core.py", line 134, in setup
    ok = dist.parse_command_line()
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/dist.py", line 707, in parse_command_line
    result = _Distribution.parse_command_line(self)
  File "/usr/lib/python3.10/distutils/dist.py", line 480, in parse_command_line
    if self.handle_display_options(option_order):
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/dist.py", line 1092, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
  File "/usr/lib/python3.10/distutils/dist.py", line 683, in handle_display_options
    self.print_commands()
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/dist.py", line 843, in print_commands
    cmdclass = ep.resolve()
  File "/home/munim/.local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/munim/.local/lib/python3.10/site-packages/setuptools/command/bdist_wininst.py", line 1, in <module>
    import distutils.command.bdist_wininst as orig
ModuleNotFoundError: **No module named 'distutils.command.bdist_wininst'**

bhung-bdai, 2023-08-15:
Verify you have those installed with the instructions here:
https://askubuntu.com/questions/1239829/modulenotfounderror-no-module-named-distutils-util

bhung-bdai, 2023-08-22:
Apologies for the late notice. It looks like you need a new version of setuptools, which you can install using: 
`sudo pip3 install --force-reinstall -v "setuptools==59.6.0"` if the reinstallation didn't work.

closed 2023-08-28 by bhung-bdai