rai-opensource/spot_ros2 #128: /list_world_objects fails when sent a specific object_type or blank

zmk5, 2023-08-15:
Hi everyone,

We've been adding a ROS 2 service client to connect to the `/list_world_objects` server, however, we've been running into a major error when attempting to request just the fiducial markers it is able to detect. The code snippet below is what we are attempting to run in our code:

```python
from bosdyn.api import world_object_pb2
from bosdyn_msgs.msg import WorldObjectType
from spot_msgs.srv import ListWorldObjects

...

request = ListWorldObjects.Request()
fiducial = WorldObjectType()
fiducial.value = world_object_pb2.WORLD_OBJECT_APRILTAG
request.request.object_type = [fiducial]
found_fiducials = robot.command("list", request=request)

...
```

where `robot.command()` is a modification that leads to the following call:

```python
...

def command(action: str, **kwargs) -> Any:
    if command == "list":
        if kwargs.get("request") is not None:
            request = kwargs["request"]
        else:
            request = ListWorldObjects.Request()
            future = self._command_map[command].call_async(request)

...
```

When doing so, we consistently get this error:

[image]

We've noticed that this same error appears when launching the `spot_driver` and using the command line `ros2 service call` feature like this: `ros2 service call /list_world_objects spot_msgs/srv/ListWorldObjects`. This command sends a blank service call but keeps erroring because of a a missing `transform_snapshots` attribute. Are we doing/calling the service wrong or is this possibly an issue from the ROS 2 service? Also the request process seems very difficult, could we possibly make calling this service easier? If so I can propose a change if you would like.

OS: Ubuntu 22.04
ROS Distro: Humble
Spot Firmware: 3.3.0

amessing-bdai, 2023-08-16:
It looks like you have an old of the bosdyn_msgs package. Can you try

```
sudo dpkg -P ros-humble-bosdyn-msgs
wget -q -O /tmp/ros-humble-bosdyn-msgs_3.2.0-0jammy_amd64.deb https://github.com/bdaiinstitute/bosdyn_msgs/releases/download/v3.2.0-frametreesnapshot/ros-humble-bosdyn-msgs_3.2.0-0jammy_amd64.deb
sudo dpkg -i /tmp/ros-humble-bosdyn-msgs_3.2.0-0jammy_amd64.deb
rm /tmp/ros-humble-bosdyn-msgs_3.2.0-0jammy_amd64.deb
```

zmk5, 2023-08-18:
You are absolutely correct, I need to change my Dockerfile to reflect the new version. Sorry for the trouble!

The other thing I wanted to ask is that `ListWorldObjects.Request().request.object_type` takes in a list of `WorldObjectType` objects, but when it is sent to the `spot_driver` it takes those and just puts them into a list of integers to send to Spot via the proto message in the `spot_wrapper`. Would it be possible to change it so it takes both a list of integers or a list of `WoldObjectType`s?

amessing-bdai, 2023-08-24:
Unfortunately not at this time. The messages are auto-generated from the protos in the spot-sdk and we don't have anyone actively working on the script right now

closed 2023-11-20 by amessing-bdai