rai-opensource/spot_ros2 #153: simple_walk_forward example is broken

ojferro, 2023-10-03:
**Issue:** Calls to constructors do not match definitions.

**Repro:** Follow installation procedure in the README, then attempt to run `spot_walk_forward.py`

--- 

e.g. `simple_walk_forward.py` crates a `TFListenerWrapper` as such:
```
self._tf_listener = TFListenerWrapper(
            "walk_forward_tf",
            wait_for_transform=[self._name + BODY_FRAME_NAME, self._name + VISION_FRAME_NAME],
        )
```

but the `TFListenerWrapper` class does not take a string as its first arg:

```
class TFListenerWrapper(object):
    def __init__(
        self,
        node: Optional[Node] = None,
        wait_for_transform: Optional[Tuple[str, str]] = None,
        cache_time_s: Optional[int] = None,
    )
```

bhung-bdai, 2023-10-03:
Hi,

Thanks for bringing this up. We've been upgrading the ROS2 tooling stack recently and may not have updated all the examples accordingly. I'll go ahead and look through the examples to see if there are other breaking changes that need to be addressed.

bhung-bdai, 2023-10-03:
This was my misunderstanding. The changes made by @mhidalgo-bdai have not been incorporated into this system at the moment. I will revert the changes for now until he has finished migrating those changes over

bhung-bdai, 2023-10-03:
Should be fixed here: https://github.com/bdaiinstitute/spot_ros2/pull/154

closed 2023-10-06 by bhung-bdai