rai-opensource/spot_ros2 #175: Spot shouldn't stand before stand is called

skpawar1305, 2023-10-28:
for now the behavior is, if we publish to cmd_vel, spot simply stands and walk, when it's in sit mode, should we add a flag for the same?

bhung-bdai, 2023-10-28:
> for now the behavior is, if we publish to cmd_vel, spot simply stands and walks, when it's in sit mode, should we add a flag for the same?

Hi,

If you command a velocity without the robot standing up, it won't be able to accomplish the movement. That's why it stands up. Do you mean putting a check to make sure the robot is standing before executing, and cancelling if the robot isn't standing? It wasn't designed with that intent.

skpawar1305, 2023-10-28:
> Hi,
>
> If you command a velocity without the robot standing up, it won't be able to accomplish the movement. That's why it stands up. Do you mean putting a check to make sure the robot is standing [...]

In spot app, it doesn't move till we make it stand first, I believe this is necessary for safety, but if you believe that it wasn't intended, then you may close this issue

bhung-bdai, 2023-10-30:
That was not the intended design behaviour I believe, but we will add it to our future documentation. Thanks for the heads up

skpawar1305, 2024-07-15:
This thing is important, another thought?

closed 2024-10-10 by tcappellari-bdai