rai-opensource/spot_ros2 #186: Colcon Build Failed for spot_description

mobrob, 2023-11-08:
--- stderr: spot_description
Traceback (most recent call last):
  File "/opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 22, in <module>

```
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:95 (message):
  execute_process(/home/nico/spot_env2/bin/python3.10
  /opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/nico/ros2_ws/src/spot_ros2/spot_description/package.xml
  /home/nico/ros2_ws/build/spot_description/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/humble/share/ament_lint_auto/cmake/ament_lint_auto_find_test_dependencies.cmake:31 (ament_package_xml)
  CMakeLists.txt:41 (ament_lint_auto_find_test_dependencies)
```
Failed   <<< spot_description [2.35s, exited with code 1]

Ubuntu 22.04
ROS Humble
virtual environment used

Codingsimon, 2023-11-12:
I had the same problem. Running `colcon build --symlink-install`  within the spot driver folder (spot_ros2) worked for me.

mobrob, 2023-11-12:
Unfortunately, I still get the same error in my case.

skpawar1305, 2023-11-14:
> --- stderr: spot_description Traceback (most recent call last): File "/opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 22, in
> 
> ```
> ModuleNotFoundError: No module named 'catkin_pkg'
> CMake Error at /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:95 (message):
>   execute_process(/home/nico/spot_env2/bin/python3.10
>   /opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
>   /home/nico/ros2_ws/src/spot_ros2/spot_description/package.xml
>   /home/nico/ros2_ws/build/spot_description/ament_cmake_core/package.cmake)
>   returned error code 1
> Call Stack (most recent call first):
>   /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
>   /opt/ros/humble/share/ament_lint_auto/cmake/ament_lint_auto_find_test_dependencies.cmake:31 (ament_package_xml)
>   CMakeLists.txt:41 (ament_lint_auto_find_test_dependencies)
> ```
> 
> Failed <<< spot_description [2.35s, exited with code 1]
> 
> Ubuntu 22.04 ROS Humble virtual environment used

I guess you can simply comment out
https://github.com/bdaiinstitute/spot_ros2/blob/d8c66b8c24a97d9a52f5444ba6f0bcd5157c1e71/spot_description/CMakeLists.txt#L41

annanya-tamu, 2024-04-01:
Was able to fix it by following @jbrodovsky's [advice](https://github.com/colcon/colcon-ros/issues/118) and running `conda install -c conda-forge catkin_pkg empy lark`.

closed 2024-06-12 by tcappellari-bdai