rai-opensource/spot_ros2 #195: Unable to create client service, robot hasn't finished booting yet

joshuaoreilly, 2023-11-16:
We're running the Spot ROS 2 driver within an Ubuntu 22 Docker container, running on an Ubuntu 20 host, and connecting to Spot with software version 3.2.2.
We're running the following commands:

```
source /ros2_ws/src/install/local_setup.bash
export SPOT_IP=<ip address of Spot>
export BOSDYN_CLIENT_PASSWORD=<password of Spot>
export BOSDYN_CLIENT_USERNAME=<username of Spot>
ros2 launch spot_driver spot_driver.launch.py config_file:=/ros2_ws/src/spot_ros2/spot_driver/config/minimal.yaml
```

where `minimal.yaml` contains:

```
/**:
  ros__parameters:
    rgb_cameras: True
```

As per [Issue 64](https://github.com/bdaiinstitute/spot_ros2/issues/64). It seems as though the driver package is successfully signing into the robot, but gets stuck:

```
root@cvgnuc1-NUC11TNHv7:/ros2_ws/src/spot_ros2/spot_driver/config# ros2 launch spot_driver [spot_driver.launch.py](http://spot_driver.launch.py/) config_file:=/ros2_ws/src/spot_ros2/spot_driver/config/config_ros2.yaml
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-11-16-11-52-37-285988-cvgnuc1-NUC11TNHv7-2914
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [spot_ros2-1]: process started with pid [2915]
[INFO] [robot_state_publisher-2]: process started with pid [2917]
[robot_state_publisher-2] [INFO] [1700135557.366502741] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1700135557.366560651] [robot_state_publisher]: got segment body
[robot_state_publisher-2] [INFO] [1700135557.366565398] [robot_state_publisher]: got segment front_left_hip
[robot_state_publisher-2] [INFO] [1700135557.366568671] [robot_state_publisher]: got segment front_left_lower_leg
[robot_state_publisher-2] [INFO] [1700135557.366571664] [robot_state_publisher]: got segment front_left_upper_leg
[robot_state_publisher-2] [INFO] [1700135557.366574331] [robot_state_publisher]: got segment front_rail
[robot_state_publisher-2] [INFO] [1700135557.366577301] [robot_state_publisher]: got segment front_right_hip
[robot_state_publisher-2] [INFO] [1700135557.366580171] [robot_state_publisher]: got segment front_right_lower_leg
[robot_state_publisher-2] [INFO] [1700135557.366583019] [robot_state_publisher]: got segment front_right_upper_leg
[robot_state_publisher-2] [INFO] [1700135557.366585680] [robot_state_publisher]: got segment rear_left_hip
[robot_state_publisher-2] [INFO] [1700135557.366588517] [robot_state_publisher]: got segment rear_left_lower_leg
[robot_state_publisher-2] [INFO] [1700135557.366591142] [robot_state_publisher]: got segment rear_left_upper_leg
[robot_state_publisher-2] [INFO] [1700135557.366593775] [robot_state_publisher]: got segment rear_rail
[robot_state_publisher-2] [INFO] [1700135557.366596399] [robot_state_publisher]: got segment rear_right_hip
[robot_state_publisher-2] [INFO] [1700135557.366599084] [robot_state_publisher]: got segment rear_right_lower_leg
[robot_state_publisher-2] [INFO] [1700135557.366601801] [robot_state_publisher]: got segment rear_right_upper_leg
[spot_ros2-1] [INFO] [1700135558.125881998] [spot_ros2]: Hi from spot_driver.
[spot_ros2-1] [INFO] [1700135558.131362485] [spot_ros2]: Starting ROS driver for Spot
[spot_ros2-1] [INFO] [1700135558.132230613] [spot_wrapper]: Initialising robot at 192.168.50.3
[spot_ros2-1] [INFO] [1700135558.132508510] [spot_wrapper]: Trying to authenticate with robot...
[spot_ros2-1] [INFO] [1700135558.909709488] [spot_wrapper]: Successfully authenticated.
[spot_ros2-1] [INFO] [1700135559.204633484] [spot_wrapper]: Creating clients...
[spot_ros2-1] [WARN] [1700135559.348581536] [spot_wrapper]: Unable to create client service: local variable 'e' referenced before assignment. This usually means the robot hasn't finished booting yet. Will wait 15 seconds and try again.
```

The final message repeats forever; this was run both after a fresh reboot, as well as an hour and a half after the robot was started.

heuristicus, 2023-11-16:
Do you have the latest wrapper code? The variable referenced before assignment issue should have been fixed a while back, and is probably hiding the actual reason for the failure to initialise.

joshuaoreilly, 2023-11-17:
Good catch, this was inside a container that wasn't keeping up with commits. Instead, I used the provided [Dockerfile](https://github.com/bdaiinstitute/spot_ros2/blob/main/.devcontainer/Dockerfile) and followed the instructions in the README to install and source the latest driver; I no longer get the above warning/hang-up, and instead get the error shown in [Issue 189](https://github.com/bdaiinstitute/spot_ros2/issues/189), so progress. Thanks!

closed 2023-11-17 by joshuaoreilly