rai-opensource/spot_ros2 #218: Cannot import choreography_sequence_pb2 from bosdyn.api.spot

kjvanteeffelen, 2023-12-22:
I followed the install instructions of the README, including the install script. Everything seems to be installed fine. However, as soon as I launch the driver I get the following error: 

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: cannot import name 'choreography_sequence_pb2' from 'bosdyn.api.spot' (/home/kees/.local/lib/python3.10/site-packages/bosdyn/api/spot/__init__.py)
```

When I run the install script again, everything surrounding the api and choreography is satisfied:

```
Requirement already satisfied: bosdyn-core==3.3.2 in /usr/local/lib/python3.10/dist-packages (from -r spot_wrapper/requirements.txt (line 1)) (3.3.2)
Requirement already satisfied: bosdyn_choreography_client==3.3.2 in /usr/local/lib/python3.10/dist-packages (from -r spot_wrapper/requirements.txt (line 2)) (3.3.2)
Requirement already satisfied: bosdyn_choreography_protos==3.3.2 in /usr/local/lib/python3.10/dist-packages (from -r spot_wrapper/requirements.txt (line 3)) (3.3.2)
Requirement already satisfied: bosdyn-client==3.3.2 in /usr/local/lib/python3.10/dist-packages (from -r spot_wrapper/requirements.txt (line 4)) (3.3.2)
Requirement already satisfied: bosdyn-mission==3.3.2 in /usr/local/lib/python3.10/dist-packages (from -r spot_wrapper/requirements.txt (line 5)) (3.3.2)
Requirement already satisfied: bosdyn-api==3.3.2 in /usr/local/lib/python3.10/dist-packages (from -r spot_wrapper/requirements.txt (line 6)) (3.3.2)
```
Is there anyone out here with the same problem, or someone that knows the cause or solution? 

Thanks!

bhung-bdai, 2024-01-08:
Apologies for the late response. I suspect it may have to do with where things are located (you may have something installed in `/usr/local/lib/python3.10/dist-packages` instead of `/usr/local/lib/python3.10/site-packages/bosdyn`, where it's trying to import from). Are you able to install these packages using `pip`?

kjvanteeffelen, 2024-01-09:
Thanks for the reply bhung. My packages are indeed located in `/distpackages`. However, I have another PC on which I installed the software a few months ago, and the packages are also located in `/distpackages`. The difference is that this PC still contains the 3.2.3 bosdyn packages instead of the latest 3.3.2. I reverted back to the 3.2.3 versions of the packages on all PCs now and the driver works. So the issue seems to be located here somewhere.

bhung-bdai, 2024-01-09:
I think your system might be importing the 3.2.3 version of the software instead of the 3.3.2 version of the `bosdyn.api.spot`. I would recommend you either separate them out or uninstall the 3.2.3 version and try again.

kjvanteeffelen, 2024-01-18:
I solved this issue, turned out I had some old version of the api packages installed in another location as well. Removing those solved this issue. Thanks, I will close it.

closed 2024-01-18 by kjvanteeffelen