rai-opensource/spot_ros2 #242: 'spot_image_publisher_node' not found on the libexec directory

orrblue, 2024-01-26:
Hi, I'm using ROS 2 Humble on a Raspberry Pi 4. I just followed the installation steps and I'm trying to run the starter code in the README.

When I run 

`source /opt/ros/humble/setup.bash` and in the ros2 workspace: `source install/local_setup.bash`

then

`ros2 launch spot_driver spot_driver.launch.py config_file:=/home/pi/Code/spot_telemetry_recorder_ros2_ws/src/spot_ros2/spot_driver/config/spot_ros_example.yaml` 

I get that spot_image_publisher_node not found:

```

[INFO] [launch]: All log files can be found below /home/pi/.ros/log/2024-01-26-10-18-37-411154-pi-desktop-184621
[INFO] [launch]: Default logging verbosity is set to INFO
MocK_Enable: False
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'spot_image_publisher_node' not found on the libexec directory '/home/pi/Code/spot_telemetry_recorder_ros2_ws/install/spot_driver_cpp/lib/spot_driver_cpp' 

```

Thank you

bhung-bdai, 2024-02-01:
This workflow is intended to be used within the docker. Are you operating inside of it?

orrblue, 2024-02-02:
Thank you for the reply @bhung-bdai. I'm not using docker and don't have experience with Docker. I was just following the README. If docker is required, it would be helpful if the docker requirement and instructions for deploying be explicitly mentioned in the readme or error message. 
In the meantime, I am avoiding the issue by simply not running the image publisher node.
Thanks for the help

khughes-bdai, 2024-04-01:
Closing this issue as the Python based image publishers have been officially deprecated with https://github.com/bdaiinstitute/spot_ros2/pull/305. Going forward, the C++ image publishers that are launched with the driver should solve this issue.

closed 2024-04-01 by khughes-bdai