rai-opensource/spot_ros2 #247: Config. file is not used to get hostname, username, or passwork when running the driver.

jacobbreen25, 2024-01-30:
When I try to run the spot_driver, despite putting a config file (which is below), I am still required to set the environmental variables to access the robot. Here is the console:

```ros2 launch spot_driver spot_driver.launch.py spot_name:="spot" config_file:="./config.yaml" depth_registered_mode:=from_spot publish_point_clouds:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/jbreen/.ros/log/2024-01-30-15-27-05-663731-csrobot-B450M-DS3H-WIFI-8776
[INFO] [launch]: Default logging verbosity is set to INFO
Failed to communicate with robot: UnableToConnectToRobotError: The robot may be offline or otherwise unreachable.
Ensure the robot is powered on and you can ping . Robot may still be booting. Will retry in 15 seconds
Failed to communicate with robot: UnableToConnectToRobotError: The robot may be offline or otherwise unreachable.
Ensure the robot is powered on and you can ping . Robot may still be booting. Will retry in 15 seconds
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```
Here is the YAML contents:
```
/**:
  ros__parameters:
    username: "user"
    password: "REDACTED"
    hostname: "192.168.80.3"
    robot_state_rate: 10.0
    metrics_rate: 0.04
    lease_rate: 1.0
    image_rate: 10.0
    auto_claim: False
    auto_power_on: False
    auto_stand: False
    deadzone: 0.05
    estop_timeout: 9.0
    start_estop: False
    preferred_odom_frame: "odom"
    cmd_duration: 0.125
    rgb_cameras: True
    initialize_spot_cam: True
```
I moved the username, password, and hostname up to the top however, this was a copy of the example code and it still didn't work

orrblue, 2024-01-30:
@jacobbreen25 Seems like we're at about the same level of familiarity with this repo. What command did you use to run spot_driver when you encountered this issue?

jacobbreen25, 2024-01-30:
@orrblue I didn't realize it didn't copy the command above. Below is my command:

```
ros2 launch spot_driver spot_driver.launch.py spot_name:="spot" config_file:=./config.yaml depth_registered_mode:="from_spot" publish_point_clouds:=true launch_rviz:=true
```

bhung-bdai, 2024-02-01:
@khughes-bdai is working on that

khughes-bdai, 2024-02-07:
Should be resolved now with this PR! https://github.com/bdaiinstitute/spot_ros2/pull/259

closed 2024-02-07 by khughes-bdai