rai-opensource/spot_ros2 #262: Can't run image publish node

orrblue, 2024-02-06:
Hi, I'm using ROS 2 Humble on a Raspberry Pi 4, and I just followed the installation steps in the README. I'm trying to make the robot publish images. I'm using `commit ac59eeb`

I tried running publish_cameras.py using both ros2 run and ros2 launch but get errors both times.
Try #1:
`ros2 run spot_driver spot_publish_cameras`
```
Traceback (most recent call last):
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/install/spot_driver/lib/spot_driver/spot_publish_cameras", line 33, in <module>
    sys.exit(load_entry_point('spot-driver==0.0.0', 'console_scripts', 'spot_publish_cameras')())
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/build/bdai_ros2_wrappers/bdai_ros2_wrappers/process.py", line 176, in __call__
    return func_taking_no_args()
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/install/spot_driver/lib/python3.10/site-packages/spot_driver/publish_cameras.py", line 205, in main
    ros_process.spin(SpotImagePublisher)
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/build/bdai_ros2_wrappers/bdai_ros2_wrappers/process.py", line 309, in spin
    process.spin(factory, *args, **kwargs)
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/build/bdai_ros2_wrappers/bdai_ros2_wrappers/scope.py", line 466, in spin
    stack.enter_context(cm)
  File "/usr/lib/python3.10/contextlib.py", line 492, in enter_context
    result = _cm_type.__enter__(cm)
  File "/usr/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/build/bdai_ros2_wrappers/bdai_ros2_wrappers/scope.py", line 299, in managed
    loaded = self.load(factory, *args, **kwargs)
  File "/home/pi/Code/spot_telemetry_recorder_ros2_ws/build/bdai_ros2_wrappers/bdai_ros2_wrappers/scope.py", line 362, in load
    node_or_graph = factory(*args, context=self._context, namespace=namespace, **kwargs)
TypeError: SpotImagePublisher.__init__() got an unexpected keyword argument 'context'
[ros2run]: Process exited with failure 1
```

Try #2
`ros2 launch spot_driver spot_image_publishers.launch.py`
and I get essentially the same error about an unexpected keyword argument.

Thank you

khughes-bdai, 2024-03-12:
HI there -- sorry for the delayed response! Just launching the standard ROS 2 driver should enable camera publishing:
```
ros2 launch spot_driver spot_driver.launch.py config_file:=your_config.yaml
```
Images will get published to 
```
/camera/back/image
/camera/frontleft/image
/camera/frontright/image
/camera/hand/image             # If your spot has an arm
/camera/left/image
/camera/right/image
```
If you want point clouds, you can add the launch argument `publish_point_clouds:=True`.

The `spot_publish_cameras` node and the `spot_image_publishers` launchfile are from an older version of our repository that used Python to publish images. The driver now uses C++ and image transport to publish images at a much faster rate, so these older files haven't been touched in a while. We are looking into removing them to avoid this confusion.

khughes-bdai, 2024-04-01:
Closing this issue as the Python based image publishers have been officially deprecated with https://github.com/bdaiinstitute/spot_ros2/pull/305. Going forward, the C++ image publishers that are launched with the driver should solve this issue.

closed 2024-04-01 by khughes-bdai