rai-opensource/spot_ros2 #280: Inverse Kinematic Client failed to authenticate robot

jacobbreen25, 2024-02-24:
While running the spot_ros2 driver using a config file, I got the following messages:

```
[ERROR] [spot_image_publisher_node-2]: process has died [pid 850059, exit code -6, cmd '/home/null/ros2_ws/install/spot_driver/lib/spot_driver/spot_image_publisher_node --ros-args -r __ns:=/uml_spot --params-file /home/null/ros2_ws/spot_config.yaml --params-file /tmp/launch_params__zq2n_p_'].
[kinematic_node-3] [ERROR] [1708809012.811125729] [uml_spot.kinematic_service]: Failed to authenticate robot: Failed to create Inverse Kinematic client.
[kinematic_node-3] terminate called after throwing an instance of 'std::runtime_error'
[kinematic_node-3]   what():  Failed to authenticate robot: Failed to create Inverse Kinematic client.
[ERROR] [kinematic_node-3]: process has died [pid 850061, exit code -6, cmd '/home/null/ros2_ws/install/spot_driver/lib/spot_driver/kinematic_node --ros-args -r __ns:=/uml_spot --params-file /home/null/ros2_ws/spot_config.yaml --params-file /tmp/launch_params_6ji2hqja'].
```

I am using the following command and config file:

```
ros2 launch spot_driver spot_driver.launch.py config_file:=/home/null/ros2_ws/spot_config.yaml spot_name:=uml_spot
```

Here is the config file:

```

/**:
  ros__parameters:
    username: "user"
    password: "exkox2qjjeqx"
    hostname: "192.168.80.3"
    robot_state_rate: 10.0
    metrics_rate: 0.04
    lease_rate: 1.0
    image_rate: 10.0
    auto_claim: False
    auto_power_on: False
    auto_stand: False
    deadzone: 0.05
    estop_timeout: 9.0
    start_estop: False
    preferred_odom_frame: "uml_spot/odom"
    cmd_duration: 0.125
    rgb_cameras: True
    initialize_spot_cam: True
```

I am also running the latest spot_ros2 driver.

khughes-bdai, 2024-02-26:
@jschornak-bdai

bhung-bdai, 2024-02-26:
What version of the Spot-SDK is your Spot robot currently running? Before 3.3.1, the IK client was not available for robots without arms. If your robot firmware is not up to date, it could be causing a crash as your laptop's API tries to create a client that isn't supported on your robot.

https://dev.bostondynamics.com/docs/release_notes.html#id4

jschornak-bdai, 2024-02-26:
@jacobbreen25 additional questions about your specific Spot, in addition to what @bhung-bdai was asking:
- Does your Spot have an arm?
- Where are you running the Spot driver nodes? What is the processor architecture of the system that's running these nodes?
  - For example, are you running these nodes on a payload computer attached to Spot, such as a [CORE I/O module](https://www.intuitive-robots.com/spot-robot-payloads-and-accessories/spot-core-i-o-payload/)?

jschornak-bdai, 2024-02-27:
@jacobbreen25 one more thing: if you run `dpkg -l spot-cpp-sdk`, what version does it report?

SDK version `3.3.0` doesn't have support for the inverse kinematics service on Spots without arms, since this is[ only added in `3.3.1` and later](https://dev.bostondynamics.com/docs/release_notes#id4). If you have `3.3.0` installed you may need to uninstall it and then download and install `3.3.2` by [running these commands from the installation script](https://github.com/bdaiinstitute/spot_ros2/blob/0f1a4cc2db66773f36ad70e21b3db9454c04e00d/install_spot_ros2.sh#L49-L52).

jacobbreen25, 2024-02-29:
The spot-cpp-sdk is running version 3.3.2. Our spot does have an arm and the spot driver node is running off of a desktop computer with x86-64 architecture.

jacobbreen25, 2024-03-06:
I believe that all of the Python libraries were updated to 3.3.2 since they were downloaded using the requirements.txt on the ros2 package but to be safe, I have updated all of them to 4.0.0 or 3.3.2 but it seems to still be happening. Here are all of my versions:

```
bosdyn-api                           4.0.0
bosdyn-choreography-client           4.0.0
bosdyn-choreography-protos           3.3.2
bosdyn-client                        4.0.0
bosdyn-core                          4.0.0
bosdyn-mission                       4.0.0
bosdyn-msgs                          3.3.2
```

jschornak-bdai, 2024-03-07:
@jacobbreen25 What version of the Spot firmware do you have installed on your Spot?

jacobbreen25, 2024-03-07:
Our Spot CORE is on version 3.2.0 and the robot itself is using version 3.2.3

jschornak-bdai, 2024-03-07:
> the robot itself is using version 3.2.3

This might be the source of the problem. The inverse kinematics service was added in 3.3.0, so the corresponding firmware for versions older than that won't support the driver's request to create the IK client.

If you have a way to update your Spot's firmware to 3.3.2 so it matches the SDK version, I bet that'll resolve this. I'm not sure how the Spot CORE version interacts with all this, though.

jschornak-bdai, 2024-03-30:
@jacobbreen25 do you have any updates on this?

jacobbreen25, 2024-03-30:
Sorry, I have been trying to get my lab to update our robot to 4.0.0 but because of some issues with arm manipulation as well as a study we are currently running relying on 3.2.0. I have access to a spot with 3.3.2 but not 4.0.0. I will let you know when we are able to update our robot and whether the issue persists

jschornak-bdai, 2024-04-02:
@jacobbreen25 could you try #329 when you have some moment? That should allow the other driver nodes to run even if the IK client couldn't be initialized, which would let you keep using firmware version `3.2.0`.

jacobbreen25, 2024-04-02:
I will be able to test this next week. I will update you on if this works then!

jschornak-bdai, 2024-04-03:
Let's keep this open until we get confirmation that #329 does indeed fix this for older firmware versions.

khughes-bdai, 2024-04-18:
HI @jacobbreen25, was this change able to resolve the error? (It's on `main` now)

jacobbreen25, 2024-04-20:
Sorry for the very late response. In a very unfortunate series of events, both of our robots were broken and have now been fixed until now. It looks like it is working so I think this issue can be closed!

PR #329 (2024-04-03): Make the IK client optional in the Spot API wrapper
## Change Overview

Spots running firmware older than 3.3.0 cannot initialize the inverse kinematics client through the Spot API. This change makes initializing the IK client optional, which lets the driver nodes that don't need it to launch and run successfully, and in turn allows the driver work for Spots running older firmware versions. This fixes #280.

I also renamed the `kinematicApi` function to `kinematicInterface` to match the pattern used for the other functions in the `SpotApi` class.

## Testing Done

- [ ] Test on a Spot with fir [... cut]
closed 2024-04-22 by tcappellari-bdai