rai-opensource/spot_ros2 #325: Clock cannot syncronize

jacobbreen25, 2024-03-30:
I am running spot on version 3.3.2 on the 330d72a branch of the spot_ros2 driver.

When I run the following command, I get a clock synchronization error.

```
ros2 launch spot_driver spot_driver.launch.py config_file:=nerveSpot.yaml spot_name:=nerve_spot
```

Here is the error I get:

```
[state_publisher_node-5] [ERROR] [1711828686.208676746] [nerve_spot.state_publisher]: Failed to get latest clock skew: Received a failure result from the TimeSyncEndpoint: 2(STATUS_MORE_SAMPLES_NEEDED): GetClockSkew: Clock synchronization not yet achieved.
[state_publisher_node-5] [ERROR] [1711828686.228131083] [nerve_spot.state_publisher]: Failed to get latest clock skew: Received a failure result from the TimeSyncEndpoint: 2(STATUS_MORE_SAMPLES_NEEDED): GetClockSkew: Clock synchronization not yet achieved.
[spot_ros2-1] Traceback (most recent call last):
[spot_ros2-1]   File "/home/jbreen/ros2_ws/install/spot_driver/lib/spot_driver/spot_ros2", line 44, in <module>
[spot_ros2-1]     from bosdyn_msgs.conversions import convert
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/bosdyn_msgs/conversions.py", line 3, in <module>
[spot_ros2-1]     import bosdyn_api_msgs.conversions as _  # noqa
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/bosdyn_api_msgs/conversions.py", line 2679, in <module>
[spot_ros2-1]     @convert.register(bosdyn_api_msgs.msg.JointCommand, bosdyn.api.basic_command_pb2.JointCommand)
[spot_ros2-1] AttributeError: module 'bosdyn.api.basic_command_pb2' has no attribute 'JointCommand'. Did you mean: 'SitCommand'?
[ERROR] [spot_ros2-1]: process has died [pid 4187, exit code 1, cmd '/home/jbreen/ros2_ws/install/spot_driver/lib/spot_driver/spot_ros2 --ros-args -r __node:=spot_ros2 -r __ns:=/nerve_spot --params-file nerveSpot.yaml --params-file /tmp/launch_params_0kxljq9s'].
[component_container_mt-7] [ERROR] [1711828687.084992784] [nerve_spot.register_node_back]: TF2 exception:
[component_container_mt-7] "nerve_spot/back_fisheye" passed to lookupTransform argument target_frame does not exist. 
[component_container_mt-7] [ERROR] [1711828687.085687986] [nerve_spot.register_node_frontleft]: TF2 exception:
[component_container_mt-7] "nerve_spot/frontleft_fisheye" passed to lookupTransform argument target_frame does not exist. 
[component_container_mt-7] [ERROR] [1711828687.085921917] [nerve_spot.register_node_frontright]: TF2 exception:
[component_container_mt-7] "nerve_spot/frontright_fisheye" passed to lookupTransform argument target_frame does not exist. 
[component_container_mt-7] [ERROR] [1711828687.086442929] [nerve_spot.register_node_left]: TF2 exception:
[component_container_mt-7] "nerve_spot/left_fisheye" passed to lookupTransform argument target_frame does not exist. 
[component_container_mt-7] [ERROR] [1711828687.087022390] [nerve_spot.register_node_right]: TF2 exception:
[component_container_mt-7] "nerve_spot/right_fisheye" passed to lookupTransform argument target_frame does not exist.
```

amessing-bdai, 2024-04-01:
@jacobbreen25 It looks like that is the commit where we upgraded to spot-sdk 4.0.0. Can you upgrade the spot-sdk and let us know if this error still persists?

jacobbreen25, 2024-04-02:
I have not upgraded to 4.0.0 but I have downgraded to checkout dc88c9c89096cb8bb85ffee8201c01781ebf21ba and the clock sync error persists but not the JointCommand error.

khughes-bdai, 2024-04-03:
Does the clock sync error happen continuously? It's expected to see a few of these messages upon initialization of the driver

khughes-bdai, 2024-04-05:
Yeah, the amount of time it takes to sync to the robot is definitely variable (more info: https://dev.bostondynamics.com/docs/concepts/base_services#time-sync). I think the JointCommand error you were seeing earlier was just from a mismatch of the spot sdk versions, so as long as the driver is able to run on this version I'm going to close this issue

closed 2024-04-05 by khughes-bdai