rai-opensource/spot_ros2 #331: handle_navigate_to in spot_ros2 given incorrect function for navigation

jacobbreen25, 2024-04-02:
I am working with the spot_ros2 driver from checkout dc88c9c89096cb8bb85ffee8201c01781ebf21ba. I am using this version due to it being compatible with 3.3.2 however, I went ahead and checked the 4.0.0 version and noticed that this issue persists there. While sending a goal to the navigate_to action server, the following error was given:

'''
[ERROR] [1712085293.141641000] [uml_spot.spot_ros2]: Error raised in execute callback: 'SpotGraphNav' object has no attribute 'navigate_to'
[spot_ros2-1] Traceback (most recent call last):
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/action/server.py", line 332, in _execute_goal
[spot_ros2-1]     execute_result = await await_or_execute(execute_callback, goal_handle)
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[spot_ros2-1]     return callback(*args)
[spot_ros2-1]   File "/home/csrobot/spot_ws/install/spot_driver/lib/spot_driver/spot_ros2", line 2597, in handle_navigate_to
[spot_ros2-1]     resp = self.spot_wrapper.spot_graph_nav.navigate_to(
[spot_ros2-1] AttributeError: 'SpotGraphNav' object has no attribute 'navigate_to'
[spot_ros2-1] [WARN] [1712085293.142679085] [uml_spot.spot_ros2]: Goal state not set, assuming aborted. Goal ID: [ 32  67 232  86  43 230  65 197 165  41 251 183 213 193 213  14]
'''

It seems that the navigate_to function was not implemented in the spot_wrapper. To get around this, I changed the navigate_to function to a combination of navigate_initial_localization from the spot wrapper and the navigate_to_existing_waypoint function, also from the spot_wrapper. This causes a leasing error (below) stating that there is no lease for resource "body". I have given up from there, since it is a bit beyond me how to fix this leasing error.

'''
[spot_ros2-1]   File "/home/csrobot/spot_ws/build/spot_wrapper/spot_wrapper/spot_graph_nav.py", line 98, in navigate_initial_localization
[spot_ros2-1]     self._lease = self._get_lease()
[spot_ros2-1]   File "/home/csrobot/spot_ws/build/spot_wrapper/spot_wrapper/spot_graph_nav.py", line 51, in _get_lease
[spot_ros2-1]     self._lease = self._lease_wallet.get_lease()
[spot_ros2-1]   File "/home/csrobot/.local/lib/python3.10/site-packages/bosdyn/client/lease.py", line 433, in get_lease
[spot_ros2-1]     return self._get_owned_lease_state_locked(resource).lease_current
[spot_ros2-1]   File "/home/csrobot/.local/lib/python3.10/site-packages/bosdyn/client/lease.py", line 476, in _get_owned_lease_state_locked
[spot_ros2-1]     lease_state = self._get_lease_state_locked(resource)
[spot_ros2-1]   File "/home/csrobot/.local/lib/python3.10/site-packages/bosdyn/client/lease.py", line 463, in _get_lease_state_locked
[spot_ros2-1]     raise NoSuchLease(resource)
[spot_ros2-1] bosdyn.client.lease.NoSuchLease: No lease for resource "body"
'''

bhung-bdai, 2024-04-03:
If the `navigate_to` function is commanding the robot to move, you will need to acquire the lease using this method: https://github.com/bdaiinstitute/spot_ros2/blob/7287d6e104fbd28188408ef64b94dc2cc719f520/spot_driver/spot_driver/spot_ros2.py#L985C1-L986C1
This is because if the lease is acquired in a different context, then getting the lease will fail. Taking the lease acquires it by force though.

TLDR: try `take_lease()` instead of `get_lease()`

bhung-bdai, 2024-04-03:
It's also unclear to me what you're trying to do. At that commit, the code had: https://github.com/bdaiinstitute/spot_wrapper/blob/3262a35bd6c6e8b290ccf5259b0bc706c5a70e11/spot_wrapper/spot_graph_nav.py#L504

Where the `_navigate_to` existed. However, I believe `_navigate_to` is functionally equivalent to `navigate_to_waypoint` 
https://github.com/bdaiinstitute/spot_wrapper/blob/3262a35bd6c6e8b290ccf5259b0bc706c5a70e11/spot_wrapper/spot_graph_nav.py#L115

bhung-bdai, 2024-04-03:
Ah, I see what the issue is now. We can push a fix to it. For your commit however, since it's in the past you might want to replace the line here:
https://github.com/bdaiinstitute/spot_ros2/blob/7287d6e104fbd28188408ef64b94dc2cc719f520/spot_driver/spot_driver/spot_ros2.py#L2743

in your version of the `spot_ros2` with `_navigate_to`

jacobbreen25, 2024-04-04:
_navigate-to has different parameters than the given navigate_to function because the given navigate_to function is also supposed to set up the localization and given graph. If I were to do it the way stated above, it would make the initial_localization_fiducials and initial_localization_waypoint parameters redundant given that they are not used, hence why I was using the navigate_initial_localization. I will try to take the lease by force to see if this solves the issue, otherwise I will try and change the navigate_to function to _navigate_to.

jacobbreen25, 2024-04-04:
Just to update, I was able to get this to work using the following code found on line 2596 in spot_ros2 

```
# run navigate_to
        if (self.spot_wrapper.spot_graph_nav.navigate_initial_localization(upload_path=goal_handle.request.upload_path, 
                initial_localization_fiducial=goal_handle.request.initial_localization_fiducial, 
                initial_localization_waypoint=goal_handle.request.initial_localization_waypoint)):
            resp = self.spot_wrapper.spot_graph_nav.navigate_to_existing_waypoint(goal_handle.request.navigate_to)
            self.run_navigate_to = False
            feedback_thread.join()
        else: 
            resp[0] = False
            resp[1] = "Unable to initialize localization"
```

and by changing line 98 in spot_graph_nav (spot_wrapper) to the following:

```self._lease = self._lease_client.take()```

I know there is probably something wrong with taking a lease like this but, for now, it has not shown any problems.

Localization is a bit funky but I am going to blame that on a bad graph.

I meant to say it in my last message, but thank you for the help in troubleshooting!

bhung-bdai, 2024-04-04:
No worries. The biggest issue you'll see here is that the `navigate_to` calls will override any current commands your robot is running. If you plan on doing a set of actions with navigations in between, I recommend you block until the actions are done before executing the next navigation or the command will forcefully move the robot to the next waypoint.

closed 2024-04-04 by bhung-bdai