rai-opensource/spot_ros2 #341: Right image rotated

annanya-tamu, 2024-04-10:
For some reason, the image I am receiving from the right camera (/camera/right/image) is coming flipped 180deg. Has anyone else experienced this?

khughes-bdai, 2024-04-10:
Hi, this is expected behavior! This is because of the way the cameras are physically mounted onto Spot. If you launch the driver with RViz, you can see that the parent tf frame of the right image (/right) is rotated almost upside-down when compared to the parent of the left image (/left). 

Here's what they should look like:
[image]

You might also notice that the frontleft and frontright cameras are rotated too, and this is because they are placed almost vertically on the front of Spot. We are working on an addition to the driver that will stitch these front two images together and give a non-rotated image.

annanya-tamu, 2024-04-10:
Ah I see, thanks for the explanation!

closed 2024-04-11 by khughes-bdai