rai-opensource/spot_ros2 #346: Packages not installing properly, can't colcon build

T-Lemmel, 2024-04-12:
**Hi, I am having a few problems trying to install Ros2 for spot, I am using an Ubuntu 22.04 and Ros2 humble

Here is what happens what i colcon build :**

colcon build --symlink-install --packages-ignore proto2ros_tests
Starting >>> bdai_ros2_wrappers
Starting >>> spot_description
Starting >>> spot_msgs
Starting >>> spot_wrapper
Starting >>> proto2ros
--- stderr: spot_wrapper                                                                                                            
/home/udf-pc1/.local/lib/python3.10/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
---
Finished <<< spot_wrapper [1.26s]
--- stderr: bdai_ros2_wrappers
/home/udf-pc1/.local/lib/python3.10/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
---
Finished <<< bdai_ros2_wrappers [1.28s]
Starting >>> spot_examples
--- stderr: spot_msgs                                                                                                                       
CMake Error at CMakeLists.txt:25 (find_package):
  By not providing "Findbosdyn_api_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "bosdyn_api_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "bosdyn_api_msgs"
  with any of the following names:

    bosdyn_api_msgsConfig.cmake
    bosdyn_api_msgs-config.cmake

  Add the installation prefix of "bosdyn_api_msgs" to CMAKE_PREFIX_PATH or
  set "bosdyn_api_msgs_DIR" to a directory containing one of the above files.
  If "bosdyn_api_msgs" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< spot_msgs [1.44s, exited with code 1]
Aborted  <<< spot_description [1.56s]                                                                                         
Aborted  <<< proto2ros [1.65s]                                                              
Aborted  <<< spot_examples [0.50s]                     

Summary: 2 packages finished [2.00s]
  1 package failed: spot_msgs
  3 packages aborted: proto2ros spot_description spot_examples
  3 packages had stderr output: bdai_ros2_wrappers spot_msgs spot_wrapper
[... 107 lines cut ...]
  File "/home/udf-pc1/Tom/ros2_ws/install/proto2ros/local/lib/python3.10/dist-packages/proto2ros/dependencies.py", line 10, in <module>
    from proto2ros.compatibility import networkx as nx
  File "/home/udf-pc1/Tom/ros2_ws/install/proto2ros/local/lib/python3.10/dist-packages/proto2ros/compatibility.py", line 10, in <module>
    import networkx
ModuleNotFoundError: No module named 'networkx'
CMake Error at /home/udf-pc1/Tom/ros2_ws/install/proto2ros/share/proto2ros/cmake/proto2ros_generate.cmake:116 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  /home/udf-pc1/Tom/ros2_ws/install/proto2ros/share/proto2ros/cmake/proto2ros_vendor_package.cmake:30 (proto2ros_generate)
  CMakeLists.txt:23 (proto2ros_vendor_package)

---
Failed   <<< bosdyn_api_msgs [22.9s, exited with code 1]

Summary: 6 packages finished [28.8s]
  2 packages failed: bosdyn_api_msgs proto2ros_tests
  5 packages had stderr output: bdai_ros2_wrappers bosdyn_api_msgs proto2ros_tests spot_examples spot_wrapper
  12 packages not processed

Does someone have an idea ? 
Thanks in advance, 
Tom

khughes-bdai, 2024-04-12:
Hello,
I think the first set of error messages came from the Boston Dynamics specific packages not being installed correctly on your system. Can you try installing them following this guide from Boston Dynamics? https://dev.bostondynamics.com/docs/python/quickstart#install-spot-python-packages . Additionally, what is the output of `dpkg -l spot-cpp-sdk`? The C++ SDK for Spot should have gotten installed with the install script, but I want to make sure it's actually there. 
For the second error, you should be able to resolve this with `pip install networkx`.

T-Lemmel, 2024-04-15:
Hello,
thanks for helping, it seems like the error was only due to networkx that i forgot to install, after building the missing packages from source, it looks ok.
Still don't know why they weren't installed through the install script though.
Anyway, it seems to work now so thanks.

closed 2024-04-15 by T-Lemmel