rai-opensource/spot_ros2 #351: ros driver does not publish

T-Lemmel, 2024-04-16:
Hi, I am having problems with the driver, it looks like there is nothing published on any topic. 

I launch it using this line :

**ros2 launch spot_driver spot_driver.launch.py config_file:=./spot_logs.yaml publish_point_clouds:=True launch_rviz:=True**
the user, password, IP... stored in my config file are good, they work with the Spot-SDK

**i get this error continuously :** 

[spot_image_publisher_node-2] [ERROR] [1713252806.883947770] [image_publisher]: Failed to get images: Failed to get images: 6(STATUS_UNSUPPORTED_PIXEL_FORMAT_REQUESTED): ImageResponse ::bosdyn::common::Status unsuccessful

In Rviz no image can be seen, same problem with Pointclouds 
[image]

**Here is the rqt_graph which looks pretty good :**

[image]

I checked using ros2 topic echo and nothing is published on cameras or depth but i get tf, odometry, battery status and some others.

Lastly, the **ros2 run spot_driver command_spot --ros-args -p command:=stand** command doesn't do anything not even an error, i just instantly get returned to the terminal

Do you have an idea ?
Thanks in advance

heuristicus, 2024-04-16:
It's likely that you need to set the `rgb_cameras` parameter to false. By default the driver assumes your robot has rgb cameras. Older versions of the robot have greyscale cameras, which can cause this error if the flag is true.

Rodhaaret, 2024-04-16:
We had the same issue back in the days and it is indeed because it assume that your Spot has rgb_camera setting it to false should fix it

khughes-bdai, 2024-04-18:
Hi @T-Lemmel, did this resolve your issue?

T-Lemmel, 2024-04-19:
hi @khughes-bdai, sorry i hadn't seen the answer, i tried it and it works :+1:

But commands like these  **ros2 run spot_driver command_spot --ros-args -p command:=stand**, still do not do anything, i do not really need it, but i wouldn't want it to indicate that something else is wrong,

khughes-bdai, 2024-04-19:
Great!
I think the `command_spot` hasn't been touched in a while -- we can look into if it can be brought up to date, or deprecated. You should be able to accomplish the same things that this node does with service calls like `ros2 service call /robotname/stand std_srvs/srv/Trigger`. You can also look into [spot_examples](https://github.com/bdaiinstitute/spot_ros2/tree/main/spot_examples) to test that everything is working okay.

T-Lemmel, 2024-04-22:
Perfect !
Thanks for the help, i'll close the issue now.

closed 2024-04-22 by T-Lemmel