rai-opensource/spot_ros2 #352: How to make Spot move slower when moving from pose to pose?

Rodhaaret, 2024-04-16:
Hey everyone!

We are currently working with Spot and trying to create some specific movement pattern in order to check if objects are located on the floor. Our challenge right now is to control the speed of Spot as it moves between the predefined points / poses. How do we achieve this?

Below i have provided a snippet of the code that creates the movement:
`
action_goals = []
proto_goals = []
rotations = []

        # original_position
        r = geometry.EulerZXY(yaw=math.radians(0), pitch=-math.radians(0))
        rotations.append(r)

        # left_down
        r = geometry.EulerZXY(yaw=math.radians(45), pitch=math.radians(45))
        rotations.append(r)

        # middle1_down 
        r = geometry.EulerZXY(yaw=math.radians(15), pitch=math.radians(45))
        rotations.append(r)

        # middle2_down 
        r = geometry.EulerZXY(yaw=-math.radians(15), pitch=math.radians(45))
        rotations.append(r)

        # right_down
        r = geometry.EulerZXY(yaw=-math.radians(45), pitch=math.radians(45))
        rotations.append(r)

        # original_position
        r = geometry.EulerZXY(yaw=math.radians(0), pitch=-math.radians(0))
        rotations.append(r)
      
        for r in rotations: 
            proto_goal = RobotCommandBuilder.synchro_stand_command(footprint_R_body=r) 
            proto_goals.append(proto_goal)
            action_goals.append(RobotCommand.Goal())
        return action_goals, proto_goals, "Look For Object"`
It is then later on executed using the following command:
                   self.robot_command_client.send_goal_and_wait(self.action_name, self.action_goals[0], 0.3)
But as mention, how do we control the speed of the movement?

heuristicus, 2024-04-16:
I think you need to use the `max_velocity` service, for which the handling code is at https://github.com/bdaiinstitute/spot_ros2/blob/d48450ec6fc670d88f92d6a8a88b65b296271b06/spot_driver/spot_driver/spot_ros2.py#L1897-L1929

khughes-bdai, 2024-04-18:
Hi @Rodhaaret, did this resolve the issue?

Rodhaaret, 2024-04-19:
Sorry for the late response. We actually discovered that our behavior tree architecture  (_which we are using with Spot_) blocked the Spot_SDK from giving responses. Fixing this also fixed the issue of it being to fast in its movement.

But thanks for the help! Will definitely look into that function in the future!