rai-opensource/spot_ros2 #361: Colcon build failed for spot_msgs

alirrezaka, 2024-04-24:
--- stderr: spot_msgs                                                                     
CMake Error at CMakeLists.txt:25 (find_package):
  By not providing "Findbosdyn_api_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "bosdyn_api_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "bosdyn_api_msgs"
  with any of the following names:

    bosdyn_api_msgsConfig.cmake
    bosdyn_api_msgs-config.cmake

  Add the installation prefix of "bosdyn_api_msgs" to CMAKE_PREFIX_PATH or
  set "bosdyn_api_msgs_DIR" to a directory containing one of the above files.
  If "bosdyn_api_msgs" provides a separate development package or SDK, be
  sure it has been installed.

Ubuntu 22.04
Ros Humble
How can i solve this error?

khughes-bdai, 2024-04-24:
Hello,
Did you try running `./install_spot_ros2.sh`, and were there any errors during this? This should install the `bosdyn_api_msgs` package.

khughes-bdai, 2024-05-07:
Hi @alirrezaka, were you able to resolve the error?

neilharrison, 2024-05-10:
Hey, I was having the exact issue - 

For me it was happening because the `ros-humble-bosdyn_msgs_4.0.0-2-jammy_amd64.run` script fails if it gets any errors, this includes errors like any dead repos in apt so if you're seeing them when you do apt update you'll need to remove them using `sudo add-apt-repository --remove ...dead.repo` 

Hope this helps!

closed 2024-06-12 by tcappellari-bdai