rai-opensource/spot_ros2 #374: frame tree not accessible ?

T-Lemmel, 2024-05-06:
Hello, I am trying to see the frame tree that is constructed when launching the spot_driver using **tf2_tools view_frames.**
But when doing that the .pdf file never appears and the .gv file looks not completely created (see below) : 

[image]

**when running the view frame command on something else then the spot it works :** 

[image]

**Here is what is typed :** 

ros2 launch spot_driver spot_driver.launch.py config_file:=./spot_logs.yaml launch_rviz:=true publish_point_clouds:=true

**and then, in another terminal :** 

ros2 run tf2_tools view_frames

**In the terminal where the driver is running i see this :** 

[component_container_mt-9] [ERROR] [1714995825.069386842] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995824.666914 but the latest data is at time 1714995824.466115, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995825.069465228] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995824.866963 but the latest data is at time 1714995824.466115, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995825.253745493] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995825.000325 but the latest data is at time 1714995824.466115, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995825.424492963] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995825.200372 but the latest data is at time 1714995824.466115, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995825.598668186] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995825.400447 but the latest data is at time 1714995824.466115, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995825.823998982] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995825.533835 but the latest data is at time 1714995825.533086, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995825.984311577] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995825.800542 but the latest data is at time 1714995825.533086, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995826.169580275] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995825.933928 but the latest data is at time 1714995825.533086, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995826.381804667] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995826.133955 but the latest data is at time 1714995825.533086, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995826.580817505] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995826.334023 but the latest data is at time 1714995825.533086, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995826.763467599] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995826.534022 but the latest data is at time 1714995825.533086, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995826.994908813] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995826.734069 but the latest data is at time 1714995826.599954, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995827.216300968] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995826.934116 but the latest data is at time 1714995826.599954, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995827.410040693] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995827.134191 but the latest data is at time 1714995826.599954, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995827.560975756] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995827.334237 but the latest data is at time 1714995826.599954, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995827.755296063] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995827.534259 but the latest data is at time 1714995826.599954, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995828.134489495] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995827.867698 but the latest data is at time 1714995827.666928, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995828.337727530] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995828.134400 but the latest data is at time 1714995827.666928, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995828.535109950] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995828.267795 but the latest data is at time 1714995827.666928, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995828.745891365] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995828.467851 but the latest data is at time 1714995827.666928, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995828.906434689] [register_node_back]: TF2 exception:
[component_container_mt-9] The tf tree is invalid because it contains a loop.
[component_container_mt-9] Frame base_link exists with parent body.
[component_container_mt-9] Frame body exists with parent odom.
[component_container_mt-9] Frame front_rail exists with parent body.
[component_container_mt-9] Frame rear_rail exists with parent body.
[component_container_mt-9] Frame gpe exists with parent odom.
[component_container_mt-9] Frame odom exists with parent body.
[component_container_mt-9] Frame vision exists with parent body.
[component_container_mt-9] Frame flat_body exists with parent body.
[component_container_mt-9] Frame front_left_hip exists with parent body.
[... 177 lines cut ...]
[component_container_mt-9] [ERROR] [1714995829.311942086] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995829.067957 but the latest data is at time 1714995828.733794, when looking up transform from frame [right] to frame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995829.517123582] [register_node_right]: TF2 exception:
[... cut ...] ame [right_fisheye]
[component_container_mt-9] [ERROR] [1714995830.867945488] [register_node_right]: TF2 exception:
[component_container_mt-9] Lookup would require extrapolation into the future.  Requested time 1714995830.668343 but the latest data is at time 1714995829.800718, when looking up transform from frame [right] to frame [right_fisheye]

**And so on... it keeps sending the same messages**

Is there a particular way that tf is used here that would explain this ? 

Thanks in advance,
Tom

khughes-bdai, 2024-05-06:
Hmm, looks like this is the source of the loop
```
Frame body exists with parent odom.
...
Frame odom exists with parent body.
```

There were some related issues with `odom` incorrectly being a child of `body` if you launch the driver without a namespace that I thought we had resolved (see https://github.com/bdaiinstitute/spot_ros2/issues/302 and https://github.com/bdaiinstitute/spot_ros2/pull/304). 

A few questions, if you launch the driver with a namespace (`spot_name:=<something>`) does this error go away? 
Additionally, what's in your `./spot_logs.yaml ` file? Does it have `preferred_odom_frame: "odom"` or something else?

khughes-bdai, 2024-05-06:
I currently don't see this error on main while launching the driver without a namespace
```
ros2 launch spot_driver spot_driver.launch.py config_file:=spot_ros.yaml  launch_rviz:=True enable_point_clouds:=True
```

This is what my config yaml looks like
```
/**:
  ros__parameters:
    robot_state_rate: 30.0
    metrics_rate: 0.04
    lease_rate: 1.0
    image_rate: 15.0
    auto_claim: False
    auto_power_on: False
    auto_stand: True
    deadzone: 0.05
    estop_timeout: 9.0
    start_estop: False
    cmd_duration: 0.25
    username: "..."
    password: "..."
    hostname: "xx.xx.xx.xx"

```

T-Lemmel, 2024-05-07:
hello, yes i had `preferred_odom_frame: "odom"`, and launching the driver with a namespace does solve it, so **thanks !** 

On the other hand, i have something weird with the **`/robot_name/odom`** frame : 

Sometimes it appears under the robot like that : 

[image]

It sometimes does, sometimes doesn't, i haven't been able to figure out why. 
(I have nothing else running on ros2 at the same time)

khughes-bdai, 2024-05-07:
Where exactly the odom frame is placed with respect to the robot is a feature of the Boston Dynamics sdk. It's normal for it to sometimes be far away from the robot. 
I am still unsure why the tf loop error was occurring for you as I wasn't able to replicate it (maybe a clash with a background node?), but I'm glad that using the namespace resolved it! Using the namespace should also help with potential tf clashes in the future.

closed 2024-05-09 by khughes-bdai