rai-opensource/spot_ros2 #381: Make install_spot_ros2.sh automatic

huyaoyu, 2024-05-16:
Hi,

When I execute `install_spot_ros2.sh` inside a terminal, I get prompted in the form of the following

```
The following additional packages will be installed:
  libprotoc-dev ...
...
0 upgraded, 31 newly installed, 0 to remove and 64 not upgraded.
Need to get 23.0 MB/34.9 MB of archives.
After this operation, 319 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

Then a manual keyboard interaction is needed to confirm the installation. This prevents me from doing this install during a `docker build` session when I try to build a docker image. We are now doing `docker commit` instead, which is not ideal for us.

After a quick check, it is the following line that generates the prompt

```
sudo /tmp/ros-humble-bosdyn_msgs_${MSG_VERSION}-jammy_${ARCH}.run
```

Could you add an option in the above `.run` file such that the install procedure could be fully automated, without user interaction?

Thank you!

khughes-bdai, 2024-05-16:
Thanks for posting, this should be a quick fix! I'll put up a PR

khughes-bdai, 2024-05-16:
Just merged in a fix to this, feel free to re-open this issue if it doesn't work!

closed 2024-05-16 by khughes-bdai