rai-opensource/spot_ros2 #390: Image publish rate is low

PranavDhulipala, 2024-05-24:
When i try doing

ros2 topic hz /spot/camera/frontleft/image

or any image source for that instance i get a super low rate. It looks like in the image publisher code atleast the images seem to be published at 15hz, I dont have any luck trying to increase the rate, i tried changing the rate in the spot image publisher, I also tried changing the parameters in the config file that i use but with no luck.

average rate: 3.990
	min: 0.144s max: 0.947s std dev: 0.14644s window: 91
average rate: 3.899
	min: 0.144s max: 0.947s std dev: 0.14890s window: 95
average rate: 3.901
	min: 0.144s max: 0.947s std dev: 0.14674s window: 99
average rate: 3.959
	min: 0.144s max: 0.947s std dev: 0.14333s window: 105
average rate: 3.987
	min: 0.144s max: 0.947s std dev: 0.14104s window: 110
average rate: 3.928
	min: 0.144s max: 0.947s std dev: 0.14288s window: 113
average rate: 3.890
	min: 0.144s max: 0.947s std dev: 0.14354s window: 116
average rate: 3.882
	min: 0.144s max: 0.947s std dev: 0.14212s window: 120
average rate: 3.881
	min: 0.144s max: 0.947s std dev: 0.14061s window: 125
average rate: 3.914
	min: 0.144s max: 0.947s std dev: 0.13857s window: 131
average rate: 3.909
	min: 0.144s max: 0.947s std dev: 0.13984s window: 135
average rate: 3.927
	min: 0.144s max: 0.947s std dev: 0.13794s window: 140
average rate: 3.893
	min: 0.144s max: 0.947s std dev: 0.13784s window: 143
average rate: 3.869
	min: 0.144s max: 0.947s std dev: 0.13901s window: 147
average rate: 3.844
	min: 0.144s max: 0.947s std dev: 0.13784s window: 151
average rate: 3.818
	min: 0.144s max: 0.947s std dev: 0.13766s window: 154
average rate: 3.788
	min: 0.144s max: 0.947s std dev: 0.13883s window: 157
average rate: 3.703
	min: 0.144s max: 1.213s std dev: 0.15753s window: 158
	
	
I also see another error probably has nothing to do with the image publisher but i get 

[spot_ros2-1] Traceback (most recent call last):
[spot_ros2-1]   File "/home/spot_ros2/install/spot_driver/lib/spot_driver/spot_ros2", line 44, in <module>
[spot_ros2-1]     from bosdyn_msgs.conversions import convert
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/bosdyn_msgs/conversions.py", line 3, in <module>
[spot_ros2-1]     import bosdyn_api_msgs.conversions as _  # noqa
[spot_ros2-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/bosdyn_api_msgs/conversions.py", line 2679, in <module>
[spot_ros2-1]     @convert.register(bosdyn_api_msgs.msg.JointCommand, bosdyn.api.basic_command_pb2.JointCommand)
[spot_ros2-1] AttributeError: module 'bosdyn.api.basic_command_pb2' has no attribute 'JointCommand'. Did you mean: 'SitCommand'?
[ERROR] [spot_ros2-1]: process has died [pid 950233, exit code 1, cmd '/home/spot_ros2/install/spot_driver/lib/spot_driver/spot_ros2 --ros-args -r __node:=spot_ros2 -r __ns:=/spot --params-file /home/spot_ros2/src/spot_ros2/spot_driver/config/spot_ros_example.yaml --params-file /tmp/launch_params_m_4lyhzm'].

PranavDhulipala, 2024-05-27:
Thanks for the reply @khughes-bdai I am connected to the robot over wifi as it is not really feasible for me to connect through ethernet at this point. I am running the driver on my laptop and i dont see the CPU spiking when the driver is running.

khughes-bdai, 2024-06-05:
Hi @PranavDhulipala,
We've recently been able to achieve faster rates by updating some DDS settings. On main now there is a file `custom_dds_profile.xml` that increases the maximum message sizes and shared memory segment sizes for FastDDS. You can enable this new profile with `export=FASTRTPS_DEFAULT_PROFILES_FILE=<path_to_file>/custom_dds_profile.xml` and it should significantly speed up image publishing. Let me know if this works for you!

closed 2024-06-21 by tcappellari-bdai