rai-opensource/spot_ros2 #392: Critical error on spot when incorrect RobotCommand action goal is sent

nishantg96, 2024-05-29:
Hello! When I send an incorrect RobotCommand to the robot_action action server, the wrapper is unable to handle that. It tries to forward the goal to the spot and spot in turn runs into a critical error with no way to recover it but to reboot it completely. 
> #### **Cleared System Fault: A critical software error has occurred, please power cycle the robot to continue. (Code: 1)**

Here is the stack trace from the spot wrapper.
```
[spot_ros2-15] [ERROR] [1717005944.113319715] [nishant_dev_spot.spot_ros2]: Error raised in execute callback: bosdyn.api.RobotCommandFeedbackResponse (InternalServerError): Command feedback not yet received for id: 2
[spot_ros2-15] Traceback (most recent call last):
[spot_ros2-15]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/action/server.py", line 332, in _execute_goal
[spot_ros2-15]     execute_result = await await_or_execute(execute_callback, goal_handle)
[spot_ros2-15]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[spot_ros2-15]     return callback(*args)
[spot_ros2-15]   File "/home/ros2-spot-wrapper/install/bdai_ros2_wrappers/lib/python3.10/site-packages/bdai_ros2_wrappers/utilities.py", line 307, in __wrapper
[spot_ros2-15]     return func(*args, **kwargs)
[spot_ros2-15]   File "/home/ros2-spot-wrapper/install/spot_driver/lib/spot_driver/spot_ros2", line 2468, in handle_robot_command_action
[spot_ros2-15]     feedback = self._get_robot_command_feedback(goal_id)
[spot_ros2-15]   File "/home/ros2-spot-wrapper/install/spot_driver/lib/spot_driver/spot_ros2", line 2385, in _get_robot_command_feedback
[spot_ros2-15]     convert(self.spot_wrapper.get_robot_command_feedback(goal_id).feedback, feedback)
[spot_ros2-15]   File "/home/ros2-spot-wrapper/install/spot_wrapper/lib/python3.10/site-packages/spot_wrapper/wrapper.py", line 1509, in get_robot_command_feedback
[spot_ros2-15]     return self._robot_command_client.robot_command_feedback(cmd_id)
[spot_ros2-15]   File "/usr/local/lib/python3.10/dist-packages/bosdyn/client/robot_command.py", line 386, in robot_command_feedback
[spot_ros2-15]     return self.call(self._stub.RobotCommandFeedback, req, None, _robot_command_feedback_error,
[spot_ros2-15]   File "/usr/local/lib/python3.10/dist-packages/bosdyn/client/common.py", line 276, in processor
[spot_ros2-15]     return func(self, rpc_method, request, value_from_response=value_from_response,
[spot_ros2-15]   File "/usr/local/lib/python3.10/dist-packages/bosdyn/client/common.py", line 416, in call
[spot_ros2-15]     return self.handle_response(response, error_from_response, value_from_response)
[spot_ros2-15]   File "/usr/local/lib/python3.10/dist-packages/bosdyn/client/common.py", line 424, in handle_response
[spot_ros2-15]     raise exc  # pylint: disable=raising-bad-type
[spot_ros2-15] bosdyn.client.exceptions.InternalServerError: bosdyn.api.RobotCommandFeedbackResponse (InternalServerError): Command feedback not yet received for id: 2
[spot_ros2-15] [WARN] [1717005944.114506524] [nishant_dev_spot.spot_ros2]: Goal state not set, assuming aborted. Goal ID: [165 204 146  34 111 207  67  89 184  85 117 143   2 171  83 220]
```

khughes-bdai, 2024-05-29:
Hi @nishantg96,
What command were you trying to send to the robot? 
To my knowledge the `spot-sdk` doesn't have a way of validating a robot command request before it is actually sent to the robot, so it might not be possible to check this from the wrapper. (I'll dig around some more in their API and see if I can find anything, but I'm not super hopeful). 
Did you get any more details about the fault besides the error message above? (more info about faults from BD: https://dev.bostondynamics.com/docs/concepts/faults.html)

jbarry-bdai, 2024-05-29:
Can you put in a copy of the `RobotCommand` message you're sending?

nishantg96, 2024-06-11:
Hi @khughes-bdai and @jbarry-bdai. Apologies for the delayed response. To give a little more information we were sending over the goals via rclnodejs and it turns out the error occurred when the **has_field**(specifically _claw_gripper_command.trajectory.has_field_) parameter of the RobotCommand goal was incorrectly set. I would also raise the issue with BD as well since leaving the robot in an unrecoverable state shouldn't happen and the errors should be handled gracefully.

closed 2024-07-08 by khughes-bdai