rai-opensource/spot_ros2 #405: Colcon Build failes when following instructions

ryan-roche, 2024-06-11:
Using the osrf/ros-humble-desktop-full image, creating a new workspace, and following the instructions, `colcon build` will fail.

khughes-bdai, 2024-06-12:
How does `colcon build` fail? Could you provide the error message?

ryan-roche, 2024-06-13:
Certainly!

I'm attempting to set up a docker container image using `osrf/ros:humble-desktop-full` as a base and adding the driver, as the robotics lab I'm at uses containers for development environments as a standard practice. I have to manually install `python3-pip`, `curl`, and `wget` as the base image does not include those.

The only thing in my workspace directory to begin with is the `src` directory, which is empty. I then clone the spot_ros2 repository into the `src` directory so that it is at `/<workspace>/src/spot_ros2/`.

I ran the `git submodule` commands outside of the container since my SSH key doesn't carry over into the container. I then bind-mount the code into the container and continue with the instructions.

The `install_spot_ros2.sh` script appears to execute just fine.

I then `cd` back to the workspace root, source `/opt/ros/humble/setup.bash`, and run the colcon build command given in the instructions. This is the output:
```
root@d869454376de:~/test2_ws# colcon build --symlink-install --packages-ignore proto2ros_tests
Starting >>> bdai_ros2_wrappers
Starting >>> spot_description
Starting >>> spot_msgs
Starting >>> spot_wrapper
Starting >>> proto2ros
--- stderr: spot_wrapper                                                                                                                                               
/usr/local/lib/python3.10/dist-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< spot_wrapper [0.80s]
--- stderr: bdai_ros2_wrappers
/usr/local/lib/python3.10/dist-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< bdai_ros2_wrappers [0.83s]
Starting >>> spot_examples
Finished <<< spot_description [1.27s]                                                                                                       
--- stderr: spot_examples                                                                                    
/usr/local/lib/python3.10/dist-packages/setuptools/command/easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< spot_examples [0.62s]
Finished <<< proto2ros [4.62s]                                                          
Finished <<< spot_msgs [32.1s]                       
Starting >>> spot_driver
[Processing: spot_driver]                                 
Finished <<< spot_driver [42.5s]                            

Summary: 7 packages finished [1min 15s]
  3 packages had stderr output: bdai_ros2_wrappers spot_examples spot_wrapper
root@d869454376de:~/test2_ws# 
```

Weirdly enough, it seems that it worked this time??? There was `stderr` output but that was just the deprecation warning message. I don't think I did anything differently this time... maybe whatever was causing the error got fixed in a commit between now and the last time I tried it?

ryan-roche, 2024-06-13:
Either way, do you think that an official Dockerfile could be made for this driver? Developing and running ROS deployments seems to be growing in popularity rather fast, and for what it's worth (and to the best of my understanding) Docker containers seem to be the preferred method of deploying code to the Spot CORE payload.

khughes-bdai, 2024-06-14:
Glad it worked this time! (Over the last week we have been upgrading this repo for the new spot-sdk 4.0.2 release, so maybe that could have been related to the failure...?)

We do have a Dockerfile in this repo that is used for our CI tests: https://github.com/bdaiinstitute/spot_ros2/blob/main/.devcontainer/Dockerfile. Will this work for your use case?

ryan-roche, 2024-06-14:
It should work as a _base_. It's a dev container dockerfile, so it's meant to give you an environment for development, but not the actual driver. As such, it may have some tools installed for development that aren't needed to run the driver in production.

I'll tinker around with it and see if I can't get a "production" dockerfile working. If I get something working, would you like me to get in touch and send it so that it can be added to the repository?

khughes-bdai, 2024-06-14:
@ryan-roche Sure -- feel free to put up a PR if you get something that works well for you

closed 2024-06-14 by khughes-bdai