rai-opensource/spot_ros2 #412: Change SSH submodule links to HTTPS

ryan-roche, 2024-06-17:
I'm trying to automate the process of installing the driver within a Dockerfile, but I'm running into an issue with the submodule links. They're SSH links, and it fails to initialize & download the submodules due to not having an SSH key configured.

I was able to run a `sed` command within the Dockerfile to replace the GitHub SSH links with their HTTPS equivalents to anonymously download the submodules (they are public repositories, after all), but would it be possible to replace the links in the actual repo with HTTPS ones?

tcappellari-bdai, 2024-06-20:
I believe they are already HTTPS? I'm not sure why your environment would want to SSH

https://github.com/bdaiinstitute/spot_ros2/blob/be4b5e0bb892c287cb38b98eadbb37d9b1fe2e12/.gitmodules#L1-L6

ryan-roche, 2024-06-20:
Strange... I must have been mistaken.

closed 2024-06-20 by ryan-roche