rai-opensource/spot_ros2 #433: Plans for Migrating to Jazzy

nishantg96, 2024-07-18:
Hello, as indicated by the title, are there any plans to migrate the wrapper to Jazzy? We're particularly interested in the potential performance improvements when using Zenoh as the middleware. The performance gains with Zenoh could be significant and we're keen to know if this is on the roadmap or if there are any existing efforts in this direction.

nishantg96, 2025-03-05:
@tcappellari-bdai Hi there! Are there any updates regarding this? If not, can we somehow contribute to get there? 😄

khughes-bdai, 2025-03-06:
@nishantg96 Unfortunately we still don't have any immediate plans for migrating to jazzy. However, if you are interested in experimenting with zenoh, there is a version for humble available that is an option to install with our provided dockerfile (thanks @mschweig !): https://github.com/bdaiinstitute/spot_ros2?tab=readme-ov-file#experimental---rmw-zenoh-middleware

khughes-bdai, 2025-07-09:
Hi @nishantg96! RAI still doesn't have any plans for migrating to jazzy, and can't commit to actively supporting this transition now since all of our workflows still rely on humble. With this in mind, I would love to open up a `jazzy` dev branch that could be open for collaboration via users like you, but after some investigation I unfortunately think there are blocking issues with `spot_ros2`s dependencies here :( 

The first immediate problem I see is the boston dynamics C++ SDK is only officially supported on 22.04 -- https://github.com/boston-dynamics/spot-cpp-sdk/blob/master/docs/cpp/README.md#supported-platforms-and-versions. Perhaps there is a way to get around this even though it's not officially supported but it is unclear to me how much work this would entail. We bring this into `spot_ros2` via [debians](https://github.com/bdaiinstitute/spot-cpp-sdk/releases) that we generate on a fork of their repo. 

Another problem I see is our `bosdyn_msgs` debian packages will not work out with jazzy as they have humble specific dependencies. We generate these [debians](https://github.com/bdaiinstitute/bosdyn_msgs/releases/tag/5.0.0) via internal workflows which depend on the debians of `spot-cpp-sdk`. So if we can't build the C++ sdk on 24.04 then making the messages for jazzy might also be blocked.

khughes-bdai, 2025-12-17:
@taDachs this is awesome! Would love to see a docker file of how you got it working. If we can get something working with Jazzy I think that would be preferable over Kilted since it's LTS and also already supported on `synchros2` main. 

FYI we have also reached out to Boston Dynamics support to see when their libraries will officially support 22.04, but it doesn't look like there were significant changes required there?

TalSomech, 2026-10-06:
Hi , wanted to know if there were any updates on this? im currently working with taDachs mirror, but wanted to know if a normal migration is scheduled to move soon as end-life of humble is nearing

tcappellari-bdai, 2026-10-07:
Hi @TalSomech 
we don't have any concrete updates on this but we are currently working on our plans for internal migrations and our OSS repos should follow soon after.

zmk5, 2026-10-07:
@TalSomech we have been using an internal version of the dockerfile that @taDachs created for Jazzy without any issues (even with Zenoh). You can even use an LLM to make it into a bash script for a native install. 

One thing to note is that we have submitted merged PRs that addressed some of the issues that @taDachs had so you may not need all of his custom repos when attempting to install. Just keep that in mind when using it.