rai-opensource/spot_ros2 #448: Image Stitcher Crash

TyHowellWork, 2024-08-12:
Hi, the image_stitcher_node is crashing immediately following launch with the following error message: 
```
[image_stitcher_node-9] terminate called after throwing an instance of 'cv::Exception'
[image_stitcher_node-9]   what():  OpenCV(4.5.4) ./modules/stitching/src/exposure_compensate.cpp:564: error: (-215:Assertion failed) _image.type() == CV_8UC3 in function 'apply'
```
The driver is running in a docker container built using the supplied Dockerfile and this [commit](https://github.com/bdaiinstitute/spot_ros2/commit/d20830cd15b8c73452455f24db6d1ff59873c32b). I suspect the failure might be related to this Spot having the non-RGB cameras but I don't have an RGB-camera Spot to test with.

khughes-bdai, 2024-08-13:
Hi @TyHowellWork ,
You are right, it looks like there are some assumptions about RGB images baked into the image stitcher. Unfortunately it is hard for me/the team at the AI Instutute to work on resolving this as we don't have any greyscale camera Spots. If you are able to get code that works with your greyscale Spots, feel free to put up a PR, and we'd be happy to help with testing it on the RGB Spots we have here. The [spot image publisher](https://github.com/bdaiinstitute/spot_ros2/blob/main/spot_driver/src/images/spot_image_publisher.cpp) has some logic that might be helpful that uses the parameter `has_rgb_cameras`.

jefflsilveira, 2024-09-25:
I fixed the issue by 
changing
```
  const auto scene_right = cv_bridge::toCvShare(left);
  const auto scene_left = cv_bridge::toCvShare(right);
```
into
```
  const auto scene_right = cv_bridge::toCvShare(left,sensor_msgs::image_encodings::BGR8);
  const auto scene_left = cv_bridge::toCvShare(right, sensor_msgs::image_encodings::BGR8);
```

khughes-bdai, 2024-10-03:
Hi @TyHowellWork 
I just put up a PR to address the issue you found. Would you mind verifying that this branch (`khughes/greyscale_stitched_images`) gives good stitched images on your greyscale cameras? 
Thanks!

TyHowellWork, 2024-10-07:
@khughes-bdai I tested the image stitcher on our greyscale Spot and it works great other than the image is rotated 90deg counter-clockwise. I will keep looking into it.

khughes-bdai, 2024-10-07:
@TyHowellWork that is the expected behavior, that's also how the image appears on the RGB side! Thanks so much for testing

closed 2024-10-10 by tcappellari-bdai