rai-opensource/spot_ros2 #509: offering incompatible QoS when when visualizing pointcloud in rviz2

JiaheXu, 2024-10-16:
Hi BDAI,

When I try to visualize the point cloud from depth cameras, the spot driver terminal shows `New publisher discovered on topic '/depth_registered/back/points', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY`. I can see grayscale images and dept-map and pointcloud topics. Please help me fix this.

Best,
Jiahe

khughes-bdai, 2024-10-16:
You will have to change the QOS policy through the rviz drop down on this topic to "Best Effort": 

[image]

closed 2024-10-16 by khughes-bdai

tcappellari-bdai, 2024-10-16:
Are you using rviz for visualization? You should be able to edit the QoS directly in it to match