rai-opensource/spot_ros2 #519: Missing configuration file for "bosdyn"

zmk5, 2024-10-30:
When attempting to compile a new version of the code, I keep running into the following error:

```bash
5.046 --- stderr: spot_hardware_interface
5.046 CMake Error at CMakeLists.txt:26 (find_package):
5.046   By not providing "Findbosdyn.cmake" in CMAKE_MODULE_PATH this project has
5.046   asked CMake to find a package configuration file provided by "bosdyn", but
5.046   CMake did not find one.
5.046
5.046   Could not find a package configuration file provided by "bosdyn" with any
5.046   of the following names:
5.046
5.046     bosdynConfig.cmake
5.046     bosdyn-config.cmake
5.046
5.046   Add the installation prefix of "bosdyn" to CMAKE_PREFIX_PATH or set
5.046   "bosdyn_DIR" to a directory containing one of the above files.  If "bosdyn"
5.046   provides a separate development package or SDK, be sure it has been
5.046   installed.
5.046
5.046
5.046 ---
5.046 Failed   <<< spot_hardware_interface [1.55s, exited with code 1]
```

I've followed the installation directions, but this package seems to be missing. Am I missing an extra dependency for the `spot_hardware_interface` package?

zmk5, 2024-10-30:
Nevermind, I think I figured it out. The message version is "4.1.0-4" which include a `ros-humble-bosdyn-cmake-module` package but version `4.1.0` does not.

khughes-bdai, 2024-10-30:
Great! Please post again if there are more issues with installation (the bosdyn cmake module addition is a fairly new change).

TuXun1999, 2025-01-22:
Hi, could you be more specific on how you solved this issue? The message version for me is "4.1.0-4" but I am still suffering from the same issue.

zmk5, 2025-01-25:
@TuXun1999  all I do is specify that the message version is "4.1.0-4" and the SDK version is "4.1.0". Like this:

```bash
SDK_VERSION="4.1.0"
MSG_VERSION="${SDK_VERSION}-4"
```