rai-opensource/spot_ros2 #526: Can't receive depth camera pointclouds

tkirc, 2024-11-21:
I run the newest version of the spot_ros2 driver corresponding to the version 4.1.0

I see all the camera related topics:
/camera/back/camera_info
/camera/back/image
/camera/frontleft/camera_info
/camera/frontleft/image
/camera/frontright/camera_info
/camera/frontright/image
/camera/hand/camera_info
/camera/hand/image
/camera/left/camera_info
/camera/left/image
/camera/right/camera_info
/camera/right/image
/cmd_vel
/depth/back/camera_info
/depth/back/image
/depth/frontleft/camera_info
/depth/frontleft/image
/depth/frontright/camera_info
/depth/frontright/image
/depth/hand/camera_info
/depth/hand/image
/depth/left/camera_info
/depth/left/image
/depth/right/camera_info
/depth/right/image
/depth_registered/frontleft/camera_info
/depth_registered/frontleft/image
/depth_registered/frontright/camera_info
/depth_registered/frontright/image
/depth_registered/left/camera_info
/depth_registered/left/image

But with ros2 topic echo /depth/frontright/image on any camera i only get messages filled with zeros.

header:
  stamp:
    sec: 1732193652
    nanosec: 762096144
  frame_id: frontright
height: 240
width: 424
encoding: 16UC1
is_bigendian: 0
step: 848
data:
- 0
- 0
- 0
- 0
- 0
- 0

- '...'
---

And in rviz i get the following warning:
[image]

To start the driver i set the ros parameter publish_point_clouds: True
Has anyone encountered a similar issue.?

tompe17, 2024-11-22:
I think you have to publish uncompressed images to get the point cloud.  At least we do not get a point cloud if we publish compressed images.

khughes-bdai, 2024-11-22:
@tkirc, what is the full command that you are using to launch the driver?

tkirc, 2024-11-22:
@khughes-bdai, I run this command with the following .yaml file as an argument:
```ros2 launch spot_driver spot_driver.launch.py config_file:=spot_ros.yaml```

```
/**:
  ros__parameters:
    robot_state_rate: 20.0
    metrics_rate: 0.04
    lease_rate: 1.0
    image_rate: 10.0
    auto_claim: False
    auto_power_on: False
    auto_stand: False
    deadzone: 0.05
    estop_timeout: 9.0
    username: "admin"
    password: ""
    hostname: "192.168.50.3"
    start_estop: False
    preferred_odom_frame: "odom"
    async_tasks_rate: 20.0
    cmd_duration: 0.3
    rgb_cameras: True
    initialize_spot_cam: False
    launch_rviz: True
    publish_point_clouds: True

```

khughes-bdai, 2024-11-22:
Ahh, publish_point_clouds is implemented as a launch argument currently. If you modify your command to be
```ros2 launch spot_driver spot_driver.launch.py config_file:=spot_ros.yaml publish_point_clouds:=True```
(and you can also take it out of your yaml) this should fix the issue!

tkirc, 2024-11-22:
@khughes-bdai thank you for your help! I try this asap.

tkirc, 2024-11-26:
I tried to start the driver with this command:
```ros2 launch spot_driver spot_driver.launch.py config_file:=spot_ros.yaml publish_point_clouds:=True```
But i still receive  messages filled with zeros and the error in rviz on the following topics:
/depth/back/image
/depth/frontleft/image
/depth/frontright/image
/depth/hand/image
/depth/left/image
/depth/right/image
/depth_registered/frontleft/image
/depth_registered/frontright/image
/depth_registered/left/image

closed 2025-01-09 by khughes-bdai