rai-opensource/spot_ros2 #529: 4.1.0 has protobuf version error during complie in Jetson Orin

JiaheXu, 2024-11-25:
Hi BDAI,

I am trying to use Jetson Orin to control the spot. The system is on jetpack6.0 ubuntu22.04, I followed the install script, the protobuf version is 4.22.1, but I found the following error during the compling:
`--- stderr: spot_hardware_interface                                                                            
In file included from /opt/spot-cpp-sdk/include/bosdyn/client/lease/lease_keepalive.h:18,
                 from /home/developer/javis_ws/src/spot_ros2/spot_hardware_interface/include/spot_hardware_interface/spot_hardware_interface.hpp:38,
                 from /home/developer/javis_ws/src/spot_ros2/spot_hardware_interface/src/spot_hardware_interface.cpp:19:
/opt/spot-cpp-sdk/include/bosdyn/api/lease.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
`

khughes-bdai, 2024-11-25:
Hmm, did you run the install script with the `--arm64` flag? Can you post the full error message from trying to build?

In the meantime, you can also try to use the provided [Dockerfile](https://github.com/bdaiinstitute/spot_ros2/blob/main/Dockerfile) (although this has only been tested on AMD64 machines, so I'm not sure if it will work out of the box).

JiaheXu, 2024-11-25:
I have re-run the install script again, the arm64 flag is attached for sure, the error is the same: `In file included from /opt/spot-cpp-sdk/include/bosdyn/api/lease.pb.h:26,
                 from /opt/spot-cpp-sdk/include/bosdyn/client/lease/lease_keepalive.h:18,
                 from /home/developer/javis_ws/src/spot_ros2/spot_hardware_interface/include/spot_hardware_interface/spot_hardware_interface.hpp:38,
                 from /home/developer/javis_ws/src/spot_ros2/spot_hardware_interface/src/spot_hardware_interface.cpp:19:
/usr/include/google/protobuf/generated_message_table_driven.h:287:25: error: ‘MapEntryHelper’ does not name a type; did you mean ‘MapEntryLite’?
  287 |   bool operator()(const MapEntryHelper<T>& a,
      |                         ^~~~~~~~~~~~~~
      |                         MapEntryLite
/usr/include/google/protobuf/generated_message_table_driven.h:287:39: error: expected ‘,’ or ‘...’ before ‘<’ token
  287 |   bool operator()(const MapEntryHelper<T>& a,
      |                                       ^
/usr/include/google/protobuf/generated_message_table_driven.h: In member function ‘bool google::protobuf::internal::CompareMapKey::operator()(int) const’:
/usr/include/google/protobuf/generated_message_table_driven.h:289:20: error: ‘a’ was not declared in this scope
  289 |     return Compare(a.key_, b.key_);
      |                    ^
/usr/include/google/protobuf/generated_message_table_driven.h:289:28: error: ‘b’ was not declared in this scope
  289 |     return Compare(a.key_, b.key_);
      |                            ^
/usr/include/google/protobuf/generated_message_table_driven.h: In function ‘void google::protobuf::internal::MapFieldSerializer(const uint8*, google::protobuf::uint32, google::protobuf::uint32, google::protobuf::uint32, google::protobuf::io::CodedOutputStream*)’:
/usr/include/google/protobuf/generated_message_table_driven.h:300:11: error: ‘MapEntryHelper’ does not name a type; did you mean ‘MapEntryLite’?
  300 |   typedef MapEntryHelper<typename MapFieldType::EntryTypeTrait> Entry;
      |           ^~~~~~~~~~~~~~
      |           MapEntryLite
/usr/include/google/protobuf/generated_message_table_driven.h:311:7: error: ‘Entry’ was not declared in this scope; did you mean ‘MapEntry’?
  311 |       Entry map_entry(*it);
      |       ^~~~~
      |       MapEntry
/usr/include/google/protobuf/generated_message_table_driven.h:313:29: error: ‘map_entry’ was not declared in this scope; did you mean ‘MapEntry’?
  313 |       output->WriteVarint32(map_entry._cached_size_);
      |                             ^~~~~~~~~
      |                             MapEntry
/usr/include/google/protobuf/generated_message_table_driven.h:318:17: error: ‘Entry’ was not declared in this scope; did you mean ‘MapEntry’?
  318 |     std::vector<Entry> v;
      |                 ^~~~~
      |                 MapEntry
/usr/include/google/protobuf/generated_message_table_driven.h:318:22: error: template argument 1 is invalid
  318 |     std::vector<Entry> v;
      |                      ^
/usr/include/google/protobuf/generated_message_table_driven.h:318:22: error: template argument 2 is invalid
/usr/include/google/protobuf/generated_message_table_driven.h:321:9: error: request for member ‘push_back’ in ‘v’, which is of non-class type ‘int’
  321 |       v.push_back(Entry(*it));
      |         ^~~~~~~~~
/usr/include/google/protobuf/generated_message_table_driven.h:323:17: error: request for member ‘begin’ in ‘v’, which is of non-class type ‘int’
  323 |     std::sort(v.begin(), v.end(), CompareMapKey());
      |                 ^~~~~
/usr/include/google/protobuf/generated_message_table_driven.h:323:28: error: request for member ‘end’ in ‘v’, which is of non-class type ‘int’
  323 |     std::sort(v.begin(), v.end(), CompareMapKey());
      |                            ^~~
/usr/include/google/protobuf/generated_message_table_driven.h:324:27: error: request for member ‘size’ in ‘v’, which is of non-class type ‘int’
  324 |     for (int i = 0; i < v.size(); i++) {
      |                           ^~~~
/usr/include/google/protobuf/generated_message_table_driven.h:326:30: error: invalid types ‘int[int]’ for array subscript
  326 |       output->WriteVarint32(v[i]._cached_size_);
      |                              ^
/usr/include/google/protobuf/generated_message_table_driven.h:327:58: error: invalid types ‘int[int]’ for array subscript
  327 |       SerializeInternal(reinterpret_cast<const uint8*>(&v[i]), t->field_table,
      |                                                          ^
In file included from /opt/spot-cpp-sdk/include/bosdyn/api/lease.pb.h:36,
                 from /opt/spot-cpp-sdk/include/bosdyn/client/lease/lease_keepalive.h:18,
                 from /home/developer/javis_ws/src/spot_ros2/spot_hardware_interface/include/spot_hardware_interface/spot_hardware_interface.hpp:38,
                 from /home/developer/javis_ws/src/spot_ros2/spot_hardware_interface/src/spot_hardware_interface.cpp:19:
/opt/spot-cpp-sdk/include/bosdyn/api/header.pb.h: At global scope:
/opt/spot-cpp-sdk/include/bosdyn/api/header.pb.h:173:25: error: ‘bosdyn::api::RequestHeader* bosdyn::api::RequestHeader::New() const’ marked ‘final’, but is not virtual
  173 |   inline RequestHeader* New() const final {
      |                         ^~~
[... cut ...]
      |                   ^~~~~~~~~~~~~~~~~~
/opt/spot-cpp-sdk/include/bosdyn/api/header.pb.h: In member function ‘void bosdyn::api::CommonError::unsafe_arena_set_allocated_message(std::string*)’:
/opt/spot-cpp-sdk/include/bosdyn/api/header.pb.h:1023:12: error: ‘struct google::protobuf::internal::ArenaStringPtr’ has no member named ‘UnsafeArenaSetAllocated’
 1023 |   message_.UnsafeArenaSetAllocated(&::PROTOBUF_NAMESPACE_ID::internal::GetEmptyStringAlreadyInited(),`
`` 

The 4.0.2 version is fine, but 4.1.0 always fail

closed 2024-11-25 by JiaheXu