rai-opensource/spot_ros2 #533: Spot Rviz / Gazebo Simulation in ros2

timbueschinghslu, 2024-12-05:
Hello 

Does anyone have experience to launch a simulation of the spot in ros2. I am unable to find any working solution to launch the spot in rviz or gazebo? 

I would appreciate any suggestions. Thanks alot!

Tim

khughes-bdai, 2024-12-05:
Hello @timbueschinghslu ,
This repo provides launchfiles to visualize the robot in rviz, but these aren't true simulations. `ros2 launch spot_driver spot_driver.launch.py ...` will bring up an rviz visualization of the physical robot, and `ros2 launch spot_description description.launch.py ...` will let you visualize the model on its own. This repo doesn't provide any integration with gazebo.

timbueschinghslu, 2024-12-05:
Hello @khughes-bdai 

Thanks for the quick reply. Do you know of any other repo which would provide such an implementation?

timbueschinghslu, 2024-12-06:
Thanks, thats the repo I tried as well.

closed 2024-12-06 by timbueschinghslu