rai-opensource/spot_ros2 #540: teleoperate the robot

olaghattas, 2024-12-28:
Hello,

I am trying to write a ROS node that allows me to teleoperate Spot's arm. In the spot-sdk, there is an arm_wasd.py example file that uses the ArmVelocityCommand API to send end-effector velocity commands. However, I cannot find an equivalent interface in this spot_ros2 package.

Here’s what I’ve tried so far:
  1. Sending poses via the robot_command action server: I computed the displacement and attempted to send new positions as poses. Unfortunately, this did not work well for my use case.
  2. Using the robot_command service: This allows me to control the body’s velocity but does not allow me to control the velocity of the end effector.

Could you please guide me on how to achieve arm teleoperation with spot_ros2? Any examples or documentation would be greatly appreciated.

For reference, here is the link to the arm_wasd.py file in the spot-sdk: https://github.com/boston-dynamics/spot-sdk/blob/master/python/examples/arm_wasd/arm_wasd.py

Thank you for your help!

tcappellari-bdai, 2025-01-14:
Hello!

There is a `cmd_vel` topic you can publish twists to (although there will likely be some delay) or you can use `ros2_control` and our hardware interface to use joint command streaming (C++).

khughes-bdai, 2025-01-21:
[This PR](https://github.com/bdaiinstitute/spot_ros2/pull/366) i believe has some work to send velocity commands to the arm in a similar way to cmd_vel, could be a useful reference

olaghattas, 2025-01-21:
Thank you both for your responses.

I tried publishing to cmd_vel, but it doesn’t actually move the robot. I launched the driver using the command:
ros2 launch spot_driver spot_driver.launch.py
I am able to use the services and run the example scripts. Do I need to run anything else for cmd_vel to work?

Additionally, the robot I have is equipped with an arm, so cmd_vel doesn’t control the arm.

Regarding ros2_control, I noticed it doesn’t include collision avoidance which is why I was trying to teleop using the SDK.

I believe the PR you mentioned is exactly what I need. Thank you so much! I’ll look into incorporating it.

khughes-bdai, 2025-01-21:
the `cmd_vel` topic should work without having to do any setup. How are you launching the driver and how are you publishing to cmd_vel?

olaghattas, 2025-01-21:
I have the Spot login data hostname, username, and password specified as environment variables. When I run ros2 launch spot_driver spot_driver.launch.py and I can see the services and topics. 

I am using the ROS CLI to publish commands to the /cmd_vel topic with ros2 topic pub /cmd_vel. I tested with different values for the geometry_msgs/Twist message. The robot seems like it is about to move, and I can hear the motors attempting to engage, but it ultimately does not move.

khughes-bdai, 2025-01-21:
@olaghattas hmm, I am testing this on robot and experiencing the same issue. Looking into this more

olaghattas, 2025-01-23:
@khughes-bdai, I tried that today, and the issue turned out to be the frequency at which cmd_vel was being published (5 Hz). After increasing it to 10 Hz, it worked. Btw the default YAML file sets cmd_duration to 0.25.

Thank you both for your help! I can now close the issue. The cmd_vel is working, and I will use the PR for the teleop setup.