rai-opensource/spot_ros2 #542: VELODYNE PointCloud Data.

obioraodugu, 2025-01-02:
I am trying to get SLAM running on my spot. For this I need pointcloud data from the VELODYNE sensor attached. Getting this data was pretty straightforward in the ros1 wrapper but it seems the ros2 driver does not provide any topic for this. The pointcloud data provided are multiple depth info from the cameras.  I would like to know if I missed something during setup for publishing pointclouds from the VELODYNE sensor.

tcappellari-bdai, 2025-01-14:
Hello!

It looks like the `velodyne-point-cloud` service still exists in `spot_wrapper` but wasn't exposed in the ROS 2 driver. I can make a ticket to address this but we also welcome external PRs if you decide to fix it yourself. Thanks!

obioraodugu, 2025-02-13:
Would appreciate you opening a ticket as I am hitting a snag trying to fix this.

khughes-bdai, 2025-02-13:
Hi @obioraodugu , can you familiarize me with how the ROS 1 repo brings up the velodyne information, as I have not used the ROS 1 repo myself? Is it this launchfile? https://github.com/heuristicus/spot_ros/blob/master/spot_driver/launch/accessories.launch
If it is just this, it might be possible to find an equivalent launchfile provided by velodyne, and just launch this side by side with the driver. (Probably in this repository: https://github.com/ros-drivers/velodyne)

obioraodugu, 2025-02-21:
Yes that is the launch file.

khughes-bdai, 2025-04-17:
RAI doesn't have any Spot's with Velodynes, so unfortunately it's hard for us to help with getting this functionality in.

If you want to go through the ROS velodyne driver -- it looks like it requires you to launch the [velodyne driver](https://github.com/ros-drivers/velodyne/tree/humble-devel/velodyne_driver) and a node from [velodyne_pointcloud](https://github.com/ros-drivers/velodyne/tree/humble-devel/velodyne_pointcloud) to convert the raw data from the driver to a pointcloud. 

If you want to go through SpotWrapper/the spot sdk, I put up a simple branch for what should (hopefully) expose this on a topic following the same thing that the [ROS 1 repo](https://github.com/heuristicus/spot_ros/blob/1364052bbd116e65b362b2371602bd596c249f46/spot_driver/src/spot_driver/spot_ros.py#L436) does: https://github.com/bdaiinstitute/spot_ros2/tree/khughes/velodyne_test. I have no idea if it will work on robot since I don't have a velodyne, but feel free to work off of this if it's useful :)

nhewitt99, 2025-04-28:
Hi @khughes-bdai, thanks for standing up that branch! I just tested it out and it was 99% of the way to working on my Spot with an EAP 2. I'm testing the code on a laptop connected to Spot's ethernet. Could also potentially test it in an Extension if there's interest. There were a few hiccups that I would be happy to work on in a PR if y'all can give some guidance to address them:
- Typo on line 502: `velodyne_cb` should be `velodyne_callback` to match method name
- Need to specify a `self.rates["point_cloud"]` value or else no points will be received. I declared a parameter called lidar_rate with a default value of 10.0, but this was arbitrary. Will this impact the lidar's RPM? Is there a way to use the wrapper to configure the lidar's settings? (EAP 2 datasheet lists 5-20hz rotation rate so this might be configurable, but I don't see anything in a quick glance over the Spot proto definitions.)
- The point cloud message's frame_id ends up as `sensor_origin_velodyne-point-cloud` which is not in the TF tree. Is there a way to get this TF from Spot's Payload configuration and include it in an existing TF publisher?

Thanks!

khughes-bdai, 2025-04-30:
Hi @nhewitt99 thanks for the feedback! Let me fix those changes on the branch, and I can put that up for a quick PR. 

> Will the rates entry impact the lidar's RPM? Is there a way to use the wrapper to configure the lidar's settings?

I am really not sure about this, I would guess no since there doesn't seem to be any proto for this kind of request but I can't find anything about it in BD's docs. Maybe BD support would be able to help answer this 

> The point cloud message's frame_id ends up as sensor_origin_velodyne-point-cloud which is not in the TF tree

This is confusing to me since if this frame is included in BD's frame snapshot it should automatically be bridged to TF from our C++ state publishers 🤔 if it's the case that this frame is only specified in the [point cloud proto](https://dev.bostondynamics.com/protos/bosdyn/api/proto_reference#pointcloudsource) we might have to do something similar to [this](https://github.com/heuristicus/spot_ros/blob/1364052bbd116e65b362b2371602bd596c249f46/spot_driver/src/spot_driver/spot_ros.py#L1418), but it's a little awkward because would break from doing all the state publishing in C++

khughes-bdai, 2025-04-30:
@nhewitt99 i put up a draft PR with the changes. It is very likely that the way i attempted to get the static transform to the lidar frame is not going to work on hardware, but hopefully it's close. the main idea was to only publish the lidar related transforms, and not the standard odom/body/vision/camera transforms as those are already broadcasted elsewhere. 
if it doesn't immediately work, feel free to put up your own PR based off of it, and i can close the one i opened since that will probably make the debugging cycle easier 😄

nhewitt99, 2025-04-30:
Thanks for the quick turnaround! That works on our hardware! (linking PR for visibility there: #659)

[image]

There is some asymmetry: the point positions are apparently relative to Spot's odometry, and as a result sensor_origin_velodyne-point-cloud actually is identical to the odom frame. By contrast, other point sources like the depth cams are relative to a sensor frame which moves with the body of the robot. This isn't blocking anything on our end but was interesting to see. Maybe just worth noting this quirk of the EAP 2 in docs somewhere.

Feel free to ping me again if y'all need any more hardware tests for the PR.

khughes-bdai, 2025-05-01:
Hmm that's interesting, the parent of the `velodyne-point-cloud` frame is `body` right? Will add a note about that in the readme

khughes-bdai, 2025-05-05:
https://github.com/bdaiinstitute/spot_ros2/pull/659 is merged in, so I'm going to close this issue!

closed 2025-05-05 by khughes-bdai

PengzhaoWu, 2025-06-26:
@Nhewitt99 @Khughes-bdai

I installed the latest spot-ros2 container on my laptop about a week ago. After running the Spot ROS2 driver as instructed, I was able to visualize the live images in RViz2, but the lidar point cloud data was not published. The Velodyne lidar is connected to the Spot Core (legacy). My goal is to process the lidar data, stream live images, and send commands from my laptop, which communicates with the Spot via Wi-Fi.

Could you assist with troubleshooting or provide guidance on how to resolve this issue?

khughes-bdai, 2025-06-30:
@PengzhaoWu Velodyne point cloud data should be published on the `<Robot Name>/velodyne/points` topic but you need to set the `use_velodyne` param to `True` (detailed here: https://github.com/bdaiinstitute/spot_ros2/tree/main/spot_driver#spot-payload)
Does this work? If you don't see the topic come up, are there any error messages on launch?

PengzhaoWu, 2025-06-30:
Hi @khughes-bdai , thanks for your prompt response.  I just tried what you suggested as the following set up:

yaml file 
/**:
  ros__parameters:
    username: "tiger_lsu"
    password: "134567789"
    hostname: "192.168.80.3"
    client_name: "ros2_spot_client"

    cameras_used: ["frontleft", "frontright"]
    rgb_cameras: False
    stitch_front_images: False

    use_velodyne: True
    velodyne_rate: 10
    publish_point_clouds: True
------------------------------------------------------------------------
launch file:
ros2 launch spot_driver spot_image_publishers.launch.py \
  config_file:=/root/config/spot_config.yaml \
  uncompress_images:=True \
  publish_compressed_images:=False \
  stitch_front_images:=False
-----------------------------------------------------
root@docker-desktop:/ros_ws# ros2 topic list
/camera/frontleft/camera_info
/camera/frontleft/image
/camera/frontright/camera_info
/camera/frontright/image
/depth/frontleft/camera_info
/depth/frontleft/image
/depth/frontright/camera_info
/depth/frontright/image
/depth_registered/frontleft/camera_info
/depth_registered/frontleft/image
/depth_registered/frontright/camera_info
/depth_registered/frontright/image
/parameter_events
/rosout
/tf
/tf_static
---------------------------------

As you can see above, there's no velodyne topic showing up. In addition, I'm not using EAP, but JUST SPOT CORE(legacy), which is connected to the Velodyne lidar.

Looking forward to your response.

Pengzhap

PengzhaoWu, 2025-06-30:
@khughes-bdai it works. After changing to the right launch file, I can see the /velodyne/points topic. However, when I try to use rviz2 to visualize it, there's actually no data. Do you have any suggestions about this? Maybe directly connect the velodyne to spot robot, not spot core?
Looking forward to your response. Thank you again!
All the best,
Pengzhao