rai-opensource/spot_ros2 #554: Failed to send command when using ros2 control

jacobbreen25, 2025-01-13:
I am attempting to use the ros2 control interface to create a joint trajectory controller and use MoveIt to move the arm. In doing this, I keep getting the following error:

```[ros2_control_node-1] [ERROR] [1736803777.858152464] [SpotHardware]: Failed to send command: '3(Streaming for joint control failed): '```

I was also getting this error with running the default controllers.

It was working without this error earlier this morning

khughes-bdai, 2025-01-13:
Can you verify that your robot still has a valid joint level API license? Assuming you've tried rebooting it too.
Due to limitations of the Spot C++ SDK it is very difficult to get a description of why this error is happening. (It exists in their Python API for joint level control but does not appear to have been carried to C++...). One reason it could be happening is your network connection to the robot is not stable. If you are connecting to the robot over wifi, you will get better behavior over Ethernet. If the connection is bad, the joint level commands could also be expiring before they reach the robot. The default ["time to live" of the joint command is 5ms](https://github.com/bdaiinstitute/spot_ros2/blob/3469a3af62d3145fd927055a2c66740e335e9819/spot_hardware_interface/src/spot_hardware_interface.cpp#L520). You can locally try increasing this to see if it makes a difference?

khughes-bdai, 2025-01-13:
another thing I tried when debugging this issue a while back was running one of BD's examples to see if it was a robot problem or a ROS problem https://github.com/boston-dynamics/spot-cpp-sdk/blob/master/cpp/examples/joint_control/wiggle_arm_example.cpp

jacobbreen25, 2025-01-16:
I installed the joint level API license a few days ago so it should not be that. It looks like my spot-cpp-sdk did not install the examples when installed from the spot_ros2 install bash (or at least I cannot find them) so I was not able to run the wiggle arm example except for the example that could be run from ROS (This defeats the purpose of us trying it in the first place but I do not think this is a ROS2 issue given that the error comes up from the message status returned from the robot). I tried bringing the time to live up to 1000 ms (Extremely high to ensure it is not the problem) and the extrapolation time up to 100 ns and it is still happening. We may try changing the wifi being used by the robot to see if this makes a difference although the ping speed was rather fast (Could still be a firewall issue)

khughes-bdai, 2025-01-16:
Trying a different wifi sounds promising. You could also try their [same examples with the python sdk](https://github.com/boston-dynamics/spot-sdk/tree/master/python/examples/joint_control) which are simpler to set up. If this doesn't work out of the box I'd recommend reaching out to BD support

jacobbreen25, 2025-01-31:
From Boston Dynamics:

Thanks for reaching out! I am sorry to let you know that Boston Dynamics does not support the AI Institute's ROS2 wrapper. From the stack trace posted in the issue you cited, it appears as though your program did not hold the lease when commanding Spot to move. It is possible that a tablet, other device, or other program was connected to Spot and held motor power, which prevented your program from taking control. You may forcefully take the lease using the LeaseClient's `take` method, or by ensuring no other program holds the lease.

I have tried releasing the lease manually on the controller but the problem still persists.

khughes-bdai, 2025-01-31:
> I was able to get the arm moving with the spot sdk.

just to be clear, was this via successfully their wiggle arm python example? or is that also still not working (with the error you posted in the stack trace above)?

jacobbreen25, 2025-02-03:
I was able to get the wiggle_arm.py script to work. It turns out the tablet does not release the lease automatically, we had to do it manually.

khughes-bdai, 2025-02-03:
And if you release the lease from the tablet, then, and try the ros2 control example, are you still in the same situation of `Failed to send command: '3(Streaming for joint control failed)`?

jacobbreen25, 2025-02-03:
Yes, it is still running this error after manually releasing the lease from the tablet

khughes-bdai, 2025-02-04:
Hmm, I don't have a great immediate solution to this, especially given that it was working initially on your robot? There is no other error message from the hardware interface at launch?
Two ideas:
We recently updated this repo to 4.1.1, you could try installing the latest main and re-running (although this probably won't make a difference as there are no updates to the joint level api according to BD's changelog). 
You can also try BD's C++ API outside of ROS by just making the example C++ executable locally and linking it to your installed version of spot-cpp-sdk. (I attached a super hacky way that I have successfully tested this in the past if it's useful:  [test-low-level.zip](https://github.com/user-attachments/files/18661564/test-low-level.zip)). it's possible that there are some differences between the python and C++ sdk, as C++ is still considered a beta release

jacobbreen25, 2025-02-10:
It looks like the ROS2 control version of this still failing even though we updated the robot. I also built the c++ example you provided and it IS working.

khughes-bdai, 2025-02-13:
To follow up on this, I put in a ticket with BD support about how to get access to a better error message with their C++ SDK. it seems that [the underlying proto](https://dev.bostondynamics.com/protos/bosdyn/api/proto_reference#jointcontrolstreamresponse-status) should contain more descriptive information about why this is failing, but it does not seem possible to get this in C++ currently. 

Another thing that could be helpful is to take a log of the robot getting into this error state and send that to BD support, they may be able to provide more information about what is going wrong. Along with this you could consider sending them an minimally reproducible example, perhaps using this repository's dockerfile as a basis (then you can specify exactly what commands you are running in a reproducible environment). They don't support this repository but it could help to have the log corresponding to the code you are running.

Really sorry that you are still dealing with this issue, and i hope we can get it resolved soon!

khughes-bdai, 2025-02-14:
Here's the response I got from BD:

> Thanks for reaching out! I have a couple thoughts that I can offer. One thing I do wish to call out is that the C++ SDK is in beta, and is not fully supported.
Similar errors occassionally appear in the Python SDK when Spot is inundated with requests. It may help to reduce the frequency at which you are sending the requests.
Instead of getting the DebugString, you may consider reading the `message` field in the proto. This contains a human-readable description of the error.
If this persists, please capture an experiment log and share it with me. You can do this by tapping "Start Log" on the tablet, or in the Logs page in the Admin Console. After that, start your requests. When it fails, end the experiment log. ...

A log sent to them of this happening would probably be the best way to figure out the cause, but you can also try:
1. Change [the following line](https://github.com/bdaiinstitute/spot_ros2/blob/e882afc9b4a75a1630c93584b8dfecc9d21fa37e/spot_hardware_interface/src/spot_hardware_interface.cpp#L571)
```
    RCLCPP_ERROR(rclcpp::get_logger("SpotHardware"), "Failed to send command: '%s'",
                 joint_control_stream.status.DebugString().c_str());
```
to print the message field instead:
```
    RCLCPP_ERROR(rclcpp::get_logger("SpotHardware"), "Failed to send command: '%s'",
                 joint_control_stream.status.message().c_str());
```
maybe this will display something more readable?
2. [the controller manager is configured to run at 333 Hz](https://github.com/bdaiinstitute/spot_ros2/blob/e882afc9b4a75a1630c93584b8dfecc9d21fa37e/spot_ros2_control/config/spot_default_controllers_with_arm.yaml#L6). Try slowing this down, maybe to 50-100 Hz? Maybe something on your system or your robot is getting throttled.

jacobbreen25, 2025-02-18:
Looks like changing from .DebugString() to .message() did not change the failure message much if at all(`ros2_control_node-1] [ERROR] [1739904670.126278522] [SpotHardware]: Failed to send command: '3(Streaming for joint control failed): '`). 

I also brought the controller manager run rate to 50 hz and I am still getting this error. I am going to get a log of the program running and I will work on getting a reproducible dockerfile example. Thank you for your help!

jacobbreen25, 2025-02-18:
It looks like I didn't save the spot_hardware_interface.cpp file with the new message type. Still not useful as the result is less information: `[ros2_control_node-1] [ERROR] [1739916470.482845070] [SpotHardware]: Failed to send command: ''`

jacobbreen25, 2025-02-25:
I have taken a log of the issues. Where would you like me to send it. I believe in the past, we have sent it through a Dropbox link

khughes-bdai, 2025-02-25:
You can send an email to support@bostondynamics.com, they should be able to provide you with a dropbox link

jacobbreen25, 2025-03-17:
From Boston Dynamics: 
"I am sorry to let you know that I am not seeing anything in the log you provided. Given that this error is appearing when you use the ROS wrapper, and not when using the wiggle_arm example (or others) directly, I suspect this is an issue with the ROS wrapper itself. Could it be in some way confused at the fact that the JointCommand.Request message is explicitly empty and contains no actual data?"

khughes-bdai, 2025-03-17:
> Could it be in some way confused at the fact that the JointCommand.Request message is explicitly empty and contains no actual data?"
Hmm this should not be possible with our implementation. Upon initialization of the hardware interface we explicitly fill out the message with the current state of the robot, then every update simply changes the joint command values in this message. Very confusing that there was nothing helpful in the log 🤔 

We discussed this a little bit earlier in the thread, but is your connection to the robot over wifi? Internally, we only support running this  stack over an ethernet connection for the lowest possible latency and to avoid colliding with other things on the wifi. One of our users ran into this exact same error message a few days ago and it was because the connection to the robot was accidentally set to be over wifi. When we switched to a direct ethernet connection the error went away. I would be very curious if this also resolves the issue for you.

khughes-bdai, 2025-03-17:
We have other internal users of the spot joint level API, who just use the sdk directly as opposed to through ROS 2 control. They also only support running their stacks over a wired connection. Some members of these teams have done testing to see if running over wifi was feasible, but found it had a tendency to time out and kick the robot out of joint level mode (which might be what's happening in this case). Even if there's nothing else on the wifi network, i'd strongly suggest running over ethernet for more reliable results

jacobbreen25, 2025-03-21:
Looks like the issue still stands when over Ethernet.

khughes-bdai, 2025-03-27:
@jacobbreen25 So sorry for the delayed response -- @tcappellari-bdai and I have tried our best to investigate this issue but we are not sure what the root of this problem is, and we are unable to replicate it ourselves. Any time we have experienced this error message internally, it has been solved by some sort of networking re-configuration (i.e. switching from wifi to ethernet, unplugging and replugging the ethernet cable, using a different ethernet dongle, etc). Of course this still does not explain how you are able to run BD's joint level examples over this same connection, our best guess is that ROS 2 control just has a greater overhead and something either on your system or on Spot is getting throttled.
We'll continue to leave this issue open in case there are any other developments

jacobbreen25, 2025-03-27:
No problem, I understand it can be difficult to troubleshoot remotely, especially if you are not able to recreate it. I will let you know if iI find anything and if there is anything else I can provide, let me know!