rai-opensource/spot_ros2 #569: Difficulty Integrating MPPI Controller with Spot's Mobility Commands in ROS2

kabirkedia, 2025-02-04:
**Description:**  
I am working on using an MPPI (Model Predictive Path Integral) controller to send waypoints to Spot in ROS2. From my understanding, Spot requires mobility commands to be sent via actions. However, integrating actions and topics in ROS2 has proven to be tricky.  

I managed to implement a solution where waypoints are published at 10 Hz as topics, and mobility commands are sent via actions. However, I am not receiving feedback from the action server, which I need to pass to the local planner for tasks like replanning.  

**Key Issues:**  
1. Difficulty in synchronizing topics (10 Hz waypoints) with Spot's action-based mobility commands.  
2. Lack of feedback from the action server, which is critical for the local planner's functionality.  

**Question:**  
Is there a recommended or more efficient way to handle this integration? Specifically:  
- How can I ensure proper feedback from the action server?  
- Are there best practices for combining topic-based waypoints with action-based mobility commands in ROS2?  

Any guidance or examples would be greatly appreciated!  

**Additional Context:**  
- Waypoints are fixed at 10 Hz and published as topics.  
- Spot's mobility commands are sent via actions.  
- Feedback from the action server is required for the local planner to make decisions (e.g., replanning).  

Best regards,  
Kabir

khughes-bdai, 2025-02-06:
Hi Kabir, How are you constructing and sending the mobility commands? If you are using the `robot_command` action interface, feeback should be published from this block of code: https://github.com/bdaiinstitute/spot_ros2/blob/5c0d74eb33a6647b62a0e17473265e6a8d33ad81/spot_driver/spot_driver/spot_ros2.py#L2039 
Depending on the specifics of your setup, it may also be possible to use the `cmd_vel` topic to navigate your robot, avoiding the need for an action altogether. Under the hood this is constructing a specific kind of mobility command only containing velocity commands.

kabirkedia, 2025-02-06:
Hey @khughes-bdai, 

Thanks for your quick reply! I wanted to share an update: I figured out a way to get Spot action feedback using the `send_goal_and_wait` command provided by `synchros2`. For now, I’m avoiding `cmd_vel` since our project doesn’t require high-velocity movements, but I appreciate the suggestion!  

The tricky part was ensuring that actions provide feedback even during timeouts. After some experimentation, I managed to solve this using `synchros2` and topic subscriptions.  

I’m considering opening a PR to demonstrate this approach. I feel like many users rely on ROS2 topics for navigation, and the current `spot_examples` don’t extensively cover how to use `synchros2` to communicate with `spot_ros2` via topics. This could be a helpful addition for others working on similar projects.  

Let me know your thoughts!

khughes-bdai, 2025-02-06:
Hi @kabirkedia, I think that would be a great addition to spot_examples if you are willing to open up a PR! we are currently working on adding some other examples there too.
In case you hand't seen it, here's another example in synchros2 of how to execute an action and wait for it to complete: https://github.com/bdaiinstitute/ros_utilities/tree/main/synchros2#invoking-an-action-synchronously
And here's another approach you can use with synchros2 for executing an action asynchronously, and listening to the feedback as it comes in: https://github.com/bdaiinstitute/ros_utilities/tree/main/synchros2#invoking-an-action-asynchronously. Maybe this could be useful for processing the feedback from your robot command

kabirkedia, 2025-02-07:
Hello thanks for providing these examples! I used these to create a subscriber and listen to feedback! I will open a PR by tomorrow!

khughes-bdai, 2025-02-13:
Closing this issue as it seems to be resolved!

closed 2025-02-13 by khughes-bdai