rai-opensource/spot_ros2 #591: Gripper camera zoom

kabirkedia, 2025-02-21:
Does the gripper camera on the arm have a zoom? The manual says so. If it does is there an api call?

khughes-bdai, 2025-02-21:
You can use the services `get_gripper_camera_parameters` to get the current configuration, and `set_gripper_camera_parameters` to change the parameters on the gripper camera. The full description of the fields of this message is [here](https://dev.bostondynamics.com/protos/bosdyn/api/proto_reference#grippercameraparams). i am not seeing anything about zoom here but you can change the resolution of the image?

kabirkedia, 2025-02-21:
Is there no option to change zoom or the focal length of the camera?

khughes-bdai, 2025-02-21:
It does not seem possible to do that with the gripper camera looking at BD's docs, it might a feature of the Spot Cam PTZ payload

kabirkedia, 2025-02-21:
Thank you so much!

closed 2025-02-21 by kabirkedia

kabirkedia, 2025-04-22:
Hi @khughes-bdai there is no topic set_gripper_camera_parameters in the driver

khughes-bdai, 2025-04-22:
Hey @kabirkedia , `get/set_gripper_camera_parameters` are services exposed here by the driver, assuming your robot has both an arm and a gripper: https://github.com/bdaiinstitute/spot_ros2/blob/018b4bfdf098801d26fab571a3adad5a2fd7d378/spot_driver/spot_driver/spot_ros2.py#L886
You can customize which cameras get published from your robot with a config file, by setting the `cameras_used` param. For an example, you can look at this one: https://github.com/bdaiinstitute/spot_ros2/blob/main/spot_driver/config/spot_ros_example.yaml
You can make a copy of this file and modify it to include the params for your specific use case. Just pass the driver the path to whichever config you want to use:
`ros2 launch spot_driver spot_driver.launch.py config_file:=<path to config>/your_config.yaml <other args>`

kabirkedia, 2025-04-22:
That's weird then, I am trying to use that but I don't see it. 
Also one more question: I am trying to increase the resolution of the gripper camera, but when I increase the resolution using the tablet, the frequency decreases. If I launch less cameras will the frequency be better?