rai-opensource/spot_ros2 #595: API Support for Gripper Camera Pointing

mmayank74567, 2025-02-25:
Hello Boston Dynamics,

On the Spot tablet, there is a feature that allows the arm to look at a selected point simply by tapping on it. I’m wondering if a similar function is available through the API. Specifically, I would like to provide Spot with a waypoint and have the gripper camera automatically orient toward it.

Is this functionality currently supported? If not, I have developed a custom implementation and would be happy to contribute it via a pull request.

CC: @kabirkedia

tcappellari-bdai, 2025-02-25:
If this is built on top of the ROS 2 driver in this repository, we are happy to take a look at a PR. If it's a change to Boston Dynamic's SDK, however, I suggest asking on their [support page](https://support.bostondynamics.com/) as the RAI Institute (formerly The AI Institute) maintains this repository and we are not directly affiliated with Boston Dynamics.

The functionality you are describing is not currently implemented in our driver but it's possible there may be something similar in BD's SDK? I'm not 100% sure either way.

Thanks!

khughes-bdai, 2025-02-26:
I think you can probably use the Gaze command for this, BD's docs on it are here: https://dev.bostondynamics.com/protos/bosdyn/api/proto_reference#bosdyn-api-GazeCommand. 
You should be able to construct the protobuf, convert it to the ROS message equivalent (there are some examples in spot_examples of how to do this), and send it as a robot_command action

closed 2025-03-06 by tcappellari-bdai